CadQuery/cadquery #9: Merge changes from dcowden/master

justbuchanan, 2018-11-22:
After sending #8, I realized that there are a lot of improvements and fixes in the main cadquery repo that haven't yet been merged into this one. While we could cherry-pick some of the most important changes, I think the best thing to do would be to merge dcowden/master into this repo so we don't miss anything. This is probably something to consider doing on a regular basis to avoid diverging too much.

If we do the merge, it will supersede #8 (which I've closed) as well as #1 and #6, in addition to adding many more improvements that have been contributed to the main repo.

I'm happy to do the merge, but @adam-urbanczyk probably has better judgement on how to handle many of the merge conflicts (there are a lot - I have gotten about half way through them). Let me know what I can do to help and if you'd like me to send a PR with the merge.

adam-urbanczyk, 2018-11-23:
That is a tricky one. At least this change is in the freecad_impl so I will need to add it by hand I think.

I'll definitely try to merge the cadquery layer.

justbuchanan, 2018-11-23:
Sounds good. I was actually surprised, but git seems to be smart enough to know that occ_impl/geom.py came from freecad_impl/geom.py, so doing a `git merge dcowden/master` does pull in the `Plane.__eq__()` changes and others into geom.py. There are many merge conflicts, but not too terrible.

adam-urbanczyk, 2018-12-21:
For reference listing missing methods form cq.Workplane

name  | 0 
-- | -- 
copy | <bound method CQ.copy of <cadquery1.cq.Wor...
deepcopy | <bound method CQ.deepcopy of <cadquery1.cq...
intersect | <bound method Workplane.intersect of <cadquery...
polarArray | <bound method Workplane.polarArray of <cadquer...
polarLine | <bound method Workplane.polarLine of <cadquery...
polarLineTo | <bound method Workplane.polarLineTo of <cadque...
radiusArc | <bound method Workplane.radiusArc of <cadquery...
sagittaArc | <bound method Workplane.sagittaArc of <cadquer...
toFreecad | <bound method CQ.toFreecad of <cadquery1.cq.Wo...

adam-urbanczyk, 2018-12-29:
I think this is solved by #65. Feel free to reopen if something pops up.

closed 2018-12-29 by adam-urbanczyk