CadQuery/cadquery #12: Update README

adam-urbanczyk, 2018-11-25:
Current README is not very informative - ideally it should explain what is the purpose of this fork of cadquery etc.

jmwright, 2019-02-15:
There was some work done on revamping the CQ 1.0 readme about a year or so ago. I was thinking about copying the README from dcowden/cadquery, adding a "why this fork" section for CQ 2.0, then just cleaning up the rest the way I did with the Sphinx docs. @adam-urbanczyk what are your thoughts?

adam-urbanczyk, 2019-02-15:
Sounds good, but I'd also propose to trim it somewhat.

jmwright, 2019-02-15:
@adam-urbanczyk Are there any sections in particular that you would like to see cut?

adam-urbanczyk, 2019-02-16:
Take a look at #99 (it is a scaffold) . NB: feel free to change it/push new commits.

closed 2019-06-05 by adam-urbanczyk