CadQuery/cadquery #13: Documentation missing

adam-urbanczyk, 2018-11-25:
We need to create documentation for the PythonOCC version of cadquery / cadquery 2

fragmuffin, 2018-11-25:
To publish multiple sphinx doc versions (like read-the-docs), we could try:
https://robpol86.github.io/sphinxcontrib-versioning/v2.2.1/index.html

jmwright, 2019-01-06:
I'm planning to make a pass through the existing `doc` directory to "fix" the FreeCAD references. It probably won't cover the entire intent of this issue, and it will probably be done slowly over the next few weeks, but it's a start.

jmwright, 2019-01-06:
@dcowden @adam-urbanczyk What role (if any) does sphinx-doc still play in our documentation? Does it compile the `doc/*.rst` files and then push them to [dcowden.github.io](http://dcowden.github.io/cadquery/)? Where does that deploy happen?

Would there be any objections to pushing the CQ 2.0 documentation to the CadQuery.github.io instead?

dcowden, 2019-01-06:
I'm guessing that the sphinx-doc isnt being generated for cq 2.0.  AFAIK dcowden.github.io is still created from builds of master on the dcowden codebase

It was a non-trivial effort to build the sphinx plugin to execute scripts and render the result-- that's the main reason i used sphinxdoc. I'd like to preserve that if at all possible.

jmwright, 2019-02-06:
There's a PR for this now: #84

jmwright, 2019-02-15:
PR #84 has been merged now, so we just have to get the docs deploying properly with GitHub pages. I've looked into it a little bit, and will try to make some time to get it working.

Peque, 2019-02-18:
Is there any advantage in using `github.io` over `readthedocs.io`?

I tried to setup Read the Docs and it seems it is working just fine:

https://cadquerytest.readthedocs.io/en/readthedocs/index.html

It was really easy to set it up (and that way you do not need to care about SCVersioning nor setting up any other deploy stage):

https://github.com/Peque/cadquery/commits/readthedocs

Some parts are broken, but it is not a Read the Docs problem, I think, as they are broken too if I try to generate the docs locally:

https://cadquerytest.readthedocs.io/en/readthedocs/selectors.html#combining-selectors

adam-urbanczyk, 2019-02-19:
@Peque thanks for looking into this! I have no opinion regarding doc hosting - rtd is fine too. @jmwright  @dcowden  any thoughts?

BTW: #101 should fix the missing svg rendering. I see some examples need fixing, nothing killing though.

jmwright, 2019-02-20:
@adam-urbanczyk I guess I don't really have a preference. RTD seems to support custom domains like GitHub Pages does, which is one of the main things I think we may want in the future, so there's no advantage/disadvantage there.

Peque, 2019-02-20:
By the way, I just finished [a Sphinx extension to visualize CadQuery parts, given a source code file](https://github.com/Peque/sphinxcadquery), in case you want to have a look at it:

It [looks like this](https://peertube.social/videos/watch/72c485e8-22f4-41c5-9258-c107b20429a6).

Current limitations/drawbacks:

- Only reads code from a source code file, not inline directive code (should be pretty easy to add that functionality if you are interested, that was simply not my use case)
- Uses STL as the intermediate representation, so the quality of the output is not great

dcowden, 2019-02-20:
uhhhh-- wow! that's amazing! so much better than the ones in the existing cq docs! the benefits of the 3d view far outweigh the triangulation quality.

Would you be willing to add ability to do inline, so that we could use this ( or a variant) to enhance the cq documentation?

Peque, 2019-02-20:
@dcowden Sure, I would be glad to. I am having fun learning about Sphinx extensions. :blush:

dcowden, 2019-02-20:
@Peque awesome. the cq basic docs would look AMAZING if they used your directive

adam-urbanczyk, 2019-02-20:
Yes +1, would be a nice addition to the docs.  

Any clue how it will behave with multiple models on one page? With x3dom I remember having some warnings about too many opengl contexts open.

adam-urbanczyk, 2019-02-20:
Docs seem to be working: https://cadquery.readthedocs.io

Peque, 2019-02-20:
@dcowden Should be working now. It did not require many changes.

@adam-urbanczyk Yeah, you are right, it seems there is a problem when loading many models. There might be [ways to fix it](https://stackoverflow.com/questions/21548247/clean-up-threejs-webgl-contexts). But I am not sure I will look at them soon.

dcowden, 2019-02-20:
one simple ( and possibly beneficial for other reasons ) solution may be to divide the samples into multiple pages. any idea how many models would be ok?

adam-urbanczyk, 2019-02-20:
All, I propose to close this issue (docs are online) and open another regarding better rendering in the docs.

Peque, 2019-02-20:
@dcowden I can visualize 8. But that may depend on the browser, have not done much testing.

For what I read, it should be possible to fix it, but I am not too familiar with JavaScript nor Thingiview.js/three.js (which the extension is using behind the scenes).

jmwright, 2019-02-20:
@adam-urbanczyk Yeah, I think you can close this. We can open a separate issue for the viewer improvements.

closed 2019-02-21 by adam-urbanczyk