CadQuery/cadquery #20: add Color property for solid objects

dcowden, 2018-12-06:
[image] **Issue by [easyw](https://github.com/easyw)**
_Saturday Aug 08, 2015 at 09:04 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/105_

----

It would be useful to add Color property for solid objects and keep color for all functions (e.g. union cut etc.)
thanks

dcowden, 2018-12-06:
[image] **Comment by [hyOzd](https://github.com/hyOzd)**
_Sunday Aug 09, 2015 at 02:39 GMT_

----

I was going to create an issue for "exporting STEP with colors". Since I know we are both here for the same thing I thought I would just add this here : ) Yeah we want to export colorfuls STEPs from cadquery.

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Sunday Aug 09, 2015 at 03:23 GMT_

----

@easyw and @hyOzd - Thanks for the ideas. The CadQuery module for FreeCAD allows you to specify the color and transparency for an object when it's rendered. However, it sounds like you are both saying that you want the color to be embedded as an intrensic part of a solid. My thinking is that the color property would only matter when exporting the solid (STLs/STEPs with color), or displaying it (in FreeCAD). Are these the only use cases you both see for it?

On the surface this doesn't sound too bad, but operations that result in multiple solids (split), or operations on compound solids might get a little interesting. I'd need to research it a bit.

dcowden, 2018-12-06:
[image] **Comment by [hyOzd](https://github.com/hyOzd)**
_Sunday Aug 09, 2015 at 04:14 GMT_

----

Let me correct; I myself only care for colors when exporting or displaying. But it would be nice to be able to fuse parts, and conserve the colors of different faces. Like freecad does;

[image]

I know this has some limitations. For example refine shape operation of FreeCAD can merge different colored faces;

[image]

But I think such glitches should be expected. User can avoid them by carefully designing the part, cutting before fusing for example;

[image]

dcowden, 2018-12-06:
[image] **Comment by [easyw](https://github.com/easyw)**
_Sunday Aug 09, 2015 at 06:46 GMT_

----

@jmwright and @hyOzd 
we are using cadquery to generate a parametric lib of ICs models for kicad...
may be you could also consider to merge hyOzd fork that seems to have some enhancement...
https://github.com/hyOzd/cadquery
thank you
Maurice

dcowden, 2018-12-06:
[image] **Comment by [hyOzd](https://github.com/hyOzd)**
_Sunday Aug 09, 2015 at 07:47 GMT_

----

@easyw Sorry for confusion. I guess you have seen some of my earlier notes somewhere : ) My repository is already merged to upstream. You should be able to run my scripts with the latest version of cadquery.

Duckle29, 2019-07-07:
How would one apply the color to the solid, and how would that tie into show_object https://github.com/CadQuery/CQ-editor/issues/53?

I'd really like this, as I use cadquery for making models for kicad :)

tolot27, 2019-07-08:
Previously, I had the some problem. Now, I'm using cqparts to create separate parts with different colors for pins, housing, screws, etc. and assemble them into one component which can be exported.

[image]

adam-urbanczyk, 2021-01-04:
Shouldn't this be closed now that Assembly exists and supports colors?

jmwright, 2021-01-04:
The blending of color is still not supported, but AFAIK that's not an in-demand feature. I think a strong case could be made for closing this. To get a simple model's color to export to STEP will it be a requirement to wrap that model in an assembly and then set it's color?

adam-urbanczyk, 2021-01-04:
Yes, one needs to use the `cq.Assembly` class to be able to specify colors (and export colored STEP).

jmwright, 2021-01-04:
Ok, I think it's fine to close this.

closed 2021-01-04 by jmwright