CadQuery/cadquery #25: toLocalCoords for shapes transforms in the wrong direction

dcowden, 2018-12-06:
[image] **Issue by [fredizzimo](https://github.com/fredizzimo)**
_Wednesday Jul 04, 2018 at 18:56 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/284_

----

Consider the following example 
```python
import cadquery as cq

box = cq.Workplane("XY").box(50, 10, 10, centered=[False, False, False])
vertex = box.vertices(">X and >Y and >Z").first().val()
workplane = cq.Workplane("XY").center(5, 10)
vector = cq.Vector(50, 10, 10)
print("The original vertex is",
    vertex.X,
    vertex.Y,
    vertex.Z)
print("The converted vertex is",
    workplane.plane.toLocalCoords(vertex).X,
    workplane.plane.toLocalCoords(vertex).Y,
    workplane.plane.toLocalCoords(vertex).Z)
print("The converterd vector is", workplane.plane.toLocalCoords(vector))
```

which outputs
```
The original vertex is 50.0 10.0 10.0
The converted vertex is 55.0 20.0 10.0
The converterd vector is Vector (45.0, 0.0, 10.0)
```

As you can see the vertex is moved in the wrong direction, while the vector version works correctly. 

The bug seems to be pretty obvious here 
```
    def toLocalCoords(self, obj):
        if isinstance(obj, Vector):
            return Vector(self.fG.multiply(obj.wrapped))
        elif isinstance(obj, cadquery.Shape):
            return obj.transformShape(self.rG)
        else:
            raise ValueError(
                "Don't know how to convert type {} to local coordinates".format(
                    type(obj))) 
```
As you can see it's using the wrong transformation matrix (`self.rG`) for the shape, while `self.fG` is used for vectors.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Thursday Jul 05, 2018 at 02:44 GMT_

----

Thanks for the report. I can't recall any reason why it's this way.

Would you mind making a PR with the fix? It looks like an oversight, if the tests pass and the examples documents look ok after the change I'm happy to merge it

Peque, 2019-12-12:
@dcowden I think this has already been fixed:

https://github.com/CadQuery/cadquery/blob/acf363d721377eb22cd39f16ce5718b9ff68f282/cadquery/occ_impl/geom.py#L586-L587

So maybe the issue can be closed. :blush:

jmwright, 2019-12-12:
@Peque That's a good catch. According to the commit history, it was fixed during the move to PythonOCC. https://github.com/CadQuery/cadquery/blame/acf363d721377eb22cd39f16ce5718b9ff68f282/cadquery/occ_impl/geom.py#L587

closed 2019-12-12 by jmwright