CadQuery/cadquery #26: New CQ GUI available

dcowden, 2018-12-06:
[image] **Issue by [adam-urbanczyk](https://github.com/adam-urbanczyk)**
_Friday Jun 08, 2018 at 13:12 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/280_

----

I made my CadQuery GUI public today. You can check it out here:

https://github.com/CadQuery/CQ-editor

[image]

dcowden, 2018-12-06:
[image] **Comment by [zignig](https://github.com/zignig)**
_Friday Jun 08, 2018 at 13:29 GMT_

----

Nice work Adam. Will have a look and install it on my machine.

adam-urbanczyk, 2019-02-23:
Closing this issue. Will add a link in the README.

closed 2019-02-23 by adam-urbanczyk