CadQuery/cadquery #28: create examples contrib repo

dcowden, 2018-12-06:
[image] **Issue by [dcowden](https://github.com/dcowden)**
_Wednesday May 23, 2018 at 15:56 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/271_

----

I agree. I'm going to create a new repo for examples, with a very open contribution policy. These will be community authored and reviewed.

I think Jeremy's suggested breakdown is perfect-- i'll move the ones listed as 'contributed examples' into the new repo.

This will provide a side benefit that the new rep will get its own README-- so i can put pictures of examples there, which are currently cluttering the main repo readme.

Suggested examples to move are:

Core Examples (stays in main CadQuery repo)
Ex001_Simple_Block.py
Ex002_Block_With_Bored_Center_Hole.py
Ex003_Pillow_Block_With_Counterbored_Holes.py
Ex004_Extruded_Cylindrical_Plate.py
Ex005_Extruded_Lines_and_Arcs.py
Ex006_Moving_the_Current_Working_Point.py
Ex007_Using_Point_Lists.py
Ex008_Polygon_Creation.py
Ex009_Polylines.py
Ex010_Defining_an_Edge_with_a_Spline.py
Ex011_Mirroring_Symmetric_Geometry.py
Ex012_Creating_Workplanes_on_Faces.py
Ex013_Locating_a_Workplane_on_a_Vertex.py
Ex014_Offset_Workplanes.py
Ex015_Rotated_Workplanes.py
Ex016_Using_Construction_Geometry.py
Ex017_Shelling_to_Create_Thin_Features.py
Ex018_Making_Lofts.py
Ex019_Counter_Sunk_Holes.py
Ex020_Rounding_Corners_with_Fillets.py
Ex021_Splitting_an_Object.py
Ex025_Revolution.py
Ex031_Sweep.py
Ex034_Sweep_Along_List_Of_Wires.py
Ex033_Shelled_Cube_Inside_Chamfer_With_Logical_Selector_Operators.py
Ex024_Using_FreeCAD_Solids_as_CQ_Objects.py
Ex026_Lego_Brick.py
Contributed Examples (Separate Repo)
Ex022_Classic_OCC_Bottle.py
Ex023_Parametric_Enclosure.py
Ex027_Remote_Enclosure.py
Ex028_Numpy.py
Ex029_Braille.py
Ex030_Panel_with_Various_Holes_for_Connector_Installation.py
Ex032_3D_Printer_Extruder_Support.py

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Tuesday Nov 27, 2018 at 11:34 GMT_

----

@dcowden Is [this](https://github.com/dcowden/cadquery-contrib) the repository you created for the contributed examples?

I'm wondering if the contrib repo should be transferred to the CadQuery org before it's filled out with content.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Tuesday Nov 27, 2018 at 11:53 GMT_

----

@jmwright Yep that's it. I agree it should be moved. I'll try to do that.

adam-urbanczyk, 2019-06-13:
Done: https://github.com/CadQuery/cadquery-contrib

closed 2019-06-13 by adam-urbanczyk