CadQuery/cadquery #31: suppress_stdout_stderr seams to suppress to much

dcowden, 2018-12-06:
[image] **Issue by [gntech](https://github.com/gntech)**
_Wednesday May 16, 2018 at 17:40 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/267_

----

When exporting a STEP-file cadquery swallows some of the things written to stdout.

Consider the following script. Run from terminal with python.

``` python
import cadquery as cq

print("before creating model")

m = cq.Workplane("XY").box(10, 10, 10)

print("after creating model and before exporting model")

with open("testfile.step", "w") as f:
    cq.exporters.exportShape(m, "STEP", f)

print("after exporting model")

```

I would expect to get to get 3 print statements but I only get the last one `after exporting model`

I suspect that this has something to with the function suppress_stdout_stderr that is designed to suppress some messages from freecad regarding the step-export.

If I run the script like so instead I get all 3 print statments but also some messages from freecad regarding step creation. Using this form of the export I bypass the suppress_stdout_stderr()

``` python
import cadquery as cq

print("before creating model")

m = cq.Workplane("XY").box(10, 10, 10)

print("after creating model and before exporting model")

m.val().exportStep("out.step")

print("after exporting model")

```

My suggestion for how to fix this issue is to contact freecad upstream to see if we can get a clean step output from freecad to get rid of suppress_stdout_stderr. Because, seriously, the suppress_stdout_stderr() is not the ideal solution. :)

dcowden, 2018-12-06:
[image] **Comment by [fragmuffin](https://github.com/fragmuffin)**
_Thursday May 17, 2018 at 06:43 GMT_

----

It seems strange that you're missing messages _before_ you run the exporter.
Is this because `print` buffers output for a short while before flushing them?
Could the fix involve flushing the buffer `print` uses?

> My suggestion for how to fix this issue is to contact freecad upstream [...] because, seriously, the suppress_stdout_stderr() is not the ideal solution. :)

totally agree :+1:. Those wacky FreeCAD people!

dcowden, 2018-12-06:
[image] **Comment by [gntech](https://github.com/gntech)**
_Thursday May 17, 2018 at 07:13 GMT_

----

> It seems strange that you're missing messages before you run the exporter.
Is this because print buffers output for a short while before flushing them?

Yes, this is what I think as well. It is bad since I want to use the print statements to show progress of the script. And the buffering makes all the print come at the very end of the script run, so much for progress. :)

dcowden, 2018-12-06:
[image] **Comment by [gntech](https://github.com/gntech)**
_Thursday May 17, 2018 at 11:08 GMT_

----

I found a possible solution to the problem that freecad buffers everything that is written to stdout until the end of execution.

`sys.stdout.flush()`

See https://forum.freecadweb.org/viewtopic.php?f=22&t=28354&hilit=Step+stdout

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Thursday May 17, 2018 at 11:21 GMT_

----

...or we could just accelerate progress to pull in @adam-urbanczyk 's OCC branch and convert this function to use OCC directly and just skip freecad.  that'd be my vote.  

Last I checked his branch had just about everything working without FreeCAD.

I think we should start pulling in his branch, and then start slowly converting CQ functions to use OCC directly, initially, using the OCC bundled with FreeCAD.

dcowden, 2018-12-06:
[image] **Comment by [fragmuffin](https://github.com/fragmuffin)**
_Thursday May 17, 2018 at 11:54 GMT_

----

@gntech `sys.stdout.flush()` worked for me!
Adding it to the start of [`suppress_stdout_stderr.__enter__`](https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/__init__.py#L44) should do the trick.

@adam-urbanczyk @dcowden 

> [adam-urbanczyk 's] branch had just about everything working without FreeCAD

That's exciting! @adam-urbanczyk is it time to start migrating `cqparts`' freecad dependencies out?

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Thursday May 17, 2018 at 12:04 GMT_

----

@adam-urbanczyk @fragmuffin  it is definitely close enough i think its time to try to make it happen. I think he reported having all of the tests passing at the time even. Our test coverage is pretty good, so I think that means we're pretty close.

I've created #268 to get the ball rolling on that.

jmwright, 2021-02-05:
This is a FreeCAD issue, so I'm closing it. I put an issue on the FreeCAD workbench repo since it is the only place the old CadQuery 1.x library is still being maintained, although at a very minimal level.

closed 2021-02-05 by jmwright