CadQuery/cadquery #32: cqgi show_object should not gobble stack traces

dcowden, 2018-12-06:
[image] **Issue by [dcowden](https://github.com/dcowden)**
_Wednesday May 09, 2018 at 19:38 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/264_

----

As noted in #263 it stinks that the cqgi show_object hides stack traces

jmwright, 2020-10-05:
I'm don't think this is an issue, at least not anymore. The traceback can be extracted from the Exception object returned by the CQGI interface. The following code snippet illustrates one method.
```python
import traceback

# Parse and build using the text
cqModel = cqgi.parse(script_text)
build_result = cqModel.build({})

if !build_result.success:
    print(''.join(traceback.format_tb(build_result.exception.__traceback__)))
```
[cq_directive](https://github.com/CadQuery/cadquery/blob/master/cadquery/cq_directive.py#L61) uses the traceback module as well, but in a way that didn't work in my context. It's still a viable option though, depending on the situation.

Anyone can feel free to re-open if they think I am in error.

jmwright, 2020-10-05:
I just noticed that some lower-level trace information is still getting truncated, so I'm reopening.
If I run the script by itself, I get the following:
```
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/ProjectOne/generate.py", line 87, in <module>
    N_1_2 = N(i, k)
  File "/home/jwright/Downloads/repos/ProjectOne/generate.py", line 71, in N
    Nl += rs_added
ValueError: operands could not be broadcast together with shapes (2,) (1,3) (2,)
```
But if I run it using CQGI, I get this:
```
  File "/home/jwright/anaconda3/envs/cadquery/lib/python3.8/site-packages/cadquery/cqgi.py", line 117, in build
    exec(c, env)
  File "<cqscript>", line 87, in <module>
  File "<cqscript>", line 71, in N
```
The CQGI trace gives you line numbers, but removes the reason for the error which makes things much harder than they have to be to debug.