CadQuery/cadquery #33: Shell Broken for Loft Objects

dcowden, 2018-12-06:
[image] **Issue by [jmwright](https://github.com/jmwright)**
_Wednesday May 09, 2018 at 14:55 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/263_

----

This issue began over on the Google Group [here](https://groups.google.com/forum/#!topic/cadquery/E0_0av34rbY). I'm posting it here so we can keep track of it.

Shelling works fine with the [cube example in the documentation](https://dcowden.github.io/cadquery/selectors.html#combining-selectors).
```python
result = cq.Workplane("XY").box(2, 2, 2) \
    .faces(">Z") \
    .shell(-0.2)
```
But no combination of face selections or shell thicknesses seems to be able to make this work.
```python
import cadquery as cq
from Helpers import show

result = cq.Workplane("front").circle(1.5) \
.workplane(offset=3.0).rect(0.75, 0.5).loft(combine=True) \
.faces("-Z").shell(-0.1)

show(result)
```
I used the old `show` function in that example because it provides a stack trace, whereas CQGI's `show_object` function does not.

Here's the main error I get.
```
BRep_API: command not doneRunning the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/jwright/.FreeCAD/Mod/cadquery_module/CQGui/Command.py", line 226, in Activated
    imp.load_source('temp_module', tempFile.name)
  File "/tmp/tmphTKlY4", line 6, in <module>
    .faces("-Z").shell(-0.1)
  File "/home/jwright/.FreeCAD/Mod/cadquery_module/Libs/cadquery/cadquery/cq.py", line 820, in shell
    s = solidRef.shell(self.objects, thickness)
  File "/home/jwright/.FreeCAD/Mod/cadquery_module/Libs/cadquery/cadquery/freecad_impl/shapes.py", line 1009, in shell
    return Shape.cast(self.wrapped.makeThickness(nativeFaces, thickness, tolerance))

BRep_API: command not done
```

dcowden, 2018-12-06:
[image] **Comment by [adam-urbanczyk](https://github.com/adam-urbanczyk)**
_Thursday May 10, 2018 at 09:46 GMT_

----

Looks like a problem with the cad kernel. Did you try with a different loft based model?

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Thursday May 10, 2018 at 12:16 GMT_

----

@adam-urbanczyk  No, I only tried this one.

adam-urbanczyk, 2019-06-13:
Seems to work in CQ2.0:
[image]

tero-vallius-jz, 2024-02-12:
Using CQ2.4.0 this problem seems still to be present. Should this ticket be reopened?

jmwright, 2024-02-12:
Maybe it's a regression in OpenCASCADE. I can also reproduce in CQ commit `551a23157026faef423e7cb7caaf6f84da91a562` (Jan 30) with OCP 7.7.2.

kaylon, 2024-04-22:
Doesn't work for me as well

```import cadquery as cq
wall_thickness = 2.0
## radius
circle_inner = 39.0 / 2.0
box_x_outer = 34.6
box_y_outer = 17.1
box_height = 30.0
funnel_height = 50.0

circle_outer = circle_inner + 2 * wall_thickness
box_x_inner = box_x_outer - 2 * wall_thickness
box_y_inner = box_y_outer - 2 * wall_thickness

result = (
    cq.Workplane("front")
    .rect(box_x_outer, box_y_outer)
    #.faces(">Z")
    #.circle(circle_outer)
    .workplane(offset=funnel_height)
    .circle(circle_outer)
    .loft(combine=True)
    .shell(-2)
)

display(result) 
```

Oddly enough it works if I use a shell between two circles the same size e.g. a cylinder, but made via the loft tool.

gnbl, 2024-05-19:
Is there a way to revert to 2.3.1 from a current Linux install via the [nightly](https://cadquery.readthedocs.io/en/latest/installation.html#linux-macos) release? I've just run into this issue getting started with CadQuery.

ccazabon, 2024-12-23:
For what it's worth, I'm running into this with CQ 2.4.0 (Linux x86_64) as well.  I can't seem to shell any lofted solid, no matter how I construct it or what params I pass to the wp.loft() method.

lukas-buergi, 2025-04-23:
I probably have the same problem but with sweeps, but I didn't investigate in detail. Shelling an extrusion worked, shelling a sweep didn't. Version: 2.5.2 from pip.