CadQuery/cadquery #34: User needs to create wires in the correct order

dcowden, 2018-12-06:
[image] **Issue by [jmwright](https://github.com/jmwright)**
_Wednesday May 02, 2018 at 16:18 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/260_

----

This was an [issue originally created](https://github.com/jmwright/cadquery-freecad-module/issues/113) by @Peque on the CadQuery FreeCAD module repo, but I think it's really an issue with the CadQuery API. This could potentially be fixed at the same time as #83 

Also see the discussion [here](https://github.com/jmwright/cadquery-freecad-module/issues/93)

dcowden, 2018-12-06:
[image] **Comment by [zwn](https://github.com/zwn)**
_Monday May 14, 2018 at 05:54 GMT_

----

There is a solution suggested in https://github.com/jmwright/cadquery-freecad-module/issues/93#issuecomment-353746977. It involves in relying on `Freecad.Face()` to create the face from a list of wires.

adam-urbanczyk, 2019-06-13:
This works correctly in CQ2.0:
[image]
CC: @Peque

closed 2019-06-13 by adam-urbanczyk