CadQuery/cadquery #36: Reliability of created solids, cleanup utilities?

dcowden, 2018-12-06:
[image] **Issue by [fragmuffin](https://github.com/fragmuffin)**
_Monday Oct 02, 2017 at 11:57 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/199_

----

I've spent the last week making threads in `cadquery`, and I feel I'm learning some hard truths about OCC's ability to reliably make a solid.

At the time of writing this, a [FreeCAD forum discussion](https://forum.freecadweb.org/viewtopic.php?f=22&t=24690) is still ongoing.

**Problems include:**
* Faces aren't connected properly (require `sewShape()` to fix)
* Face edges aren't connected (ie: not "closed")
* Face is an `Unorientable shape` (I'm still trying to figure out what this means)
* (and I'm expecting more as I continue testing)

Some [can be fixed](https://github.com/fragmuffin/cqparts/blob/develop/src/cqparts/types/threads/base.py#L239-L244), some are "show stoppers" and I'm still learning how to fix the solid I've created so I can reliably use it for boolean operations.

**A Solution?**
Ideally `cadquery`'s users shouldn't have to worry about the nuances of FreeCAD and OCC to make their creations.

Should `cadquery` host "cleanup" utilities to increase reliability of model creation?
Perhaps if reliability is paramount for a particular use then "cleanup" could be automatically performed after each operation, or for select operations, such as `sweep`, `extrude`, and other similar functions.

**Root Cause of the Problem(s)**
I'm still identifying causes, but the root cause is not `cadquery`, but OCC itself... ideally these "cleanup" algorithms would also link to a relevant bug raised in OCC so it may be promoting growth of the overall platform.

**Personal Validation**
Does the above sound right?... I feel like I'm going crazy, like perhaps I'm just using it incorrectly...
Have you experienced the same (or similar) problems I have?
If so, please list them here so we can attempt to address them all.

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Monday Oct 02, 2017 at 13:13 GMT_

----

@fragmuffin There's a little bit of precedent for this with the removeSplitter inclusion in the clean function.

https://github.com/dcowden/cadquery/blob/9f4d62ca04d385e1620332d1de9477eba83b7873/cadquery/freecad_impl/shapes.py#L967

The clean parameter is defaulted to true in functions that use `clean()`.

https://github.com/dcowden/cadquery/blob/484a1ca667b79a773c7ecc17a17b6fabb437cc06/cadquery/cq.py#L1747

The clean function could be expanded to include more operations, but it seems like some of the fix/clean operations cause more harm than good in some cases. We would need to be cautious about that.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Monday Oct 02, 2017 at 13:41 GMT_

----

@fragmuffin yes, I've had the same experience. There is a LOT of 'little details' in OCC that make simple things very hard sometimes. They vary from things that simple don't work, to maddening goose-chases like 'BRepBuilder_API messed threw and error', followed by a search of the OCC forums that results in 'oh of course-- for that you shouldnt use BRepBuilder_API, you should use BRepAlgo_API'.

as @jmwright pointed out, shapefix helps these, but in the end there is a good reason that Parasolid costs a lot of money to license, and why there are no mainstream CAD systems based on OpenCascade.

OpenCascade is absolutely, by far, the best Open source CAD kernel. But it cannot hold a candle to any of the commercially maintained kernels.

All that said, we have discussed many times the alternatives.  Initially, I created cadquery with an interface layer with the idea that it could support a reliable Parasolid backend for enterprise users.  The only other path worth considering is to tackle the challenges of building cadquery on top of a pure CSG kernel, similar to how OpenSCAD does it.  If we went this path, we'd get two properties:
    (1) rock solid stability
    (2) super duper slowness

OpenSCAD is popular because of its stability-- its actually really hard to create indeterminate geometry in a pure CSG kernel. But it's soooo slow!

I would be open to the idea of making a cadquery version implemented on top of OpenCSG, or perhaps BRL-CAD.  Performance would be the challenge.

dcowden, 2018-12-06:
[image] **Comment by [fragmuffin](https://github.com/fragmuffin)**
_Monday Oct 02, 2017 at 15:46 GMT_

----

@jmwright Oh good, there's already a place for it.

> some of the fix/clean operations cause more harm than good in some cases. We would need to be cautious about that.

I'd imagine that's a very hard line to draw; a large "grey area" stretching between 
* leave it to the mercy of OCC, and
* cadquery fixing all of OCC's issues that would become a maintenance nightmare, and likely the speed of a pure CSG kernel anyway (as @dcowden mentions)

[sigh]... open source... a love and hate affair.

@dcowden 
I'm a fan of reliability.
Do you imagine `cadquery` supporting both OpenCascade and OpenCSG (or BRL-CAD)?, or would it be a separately maintained port, like `cadquery-opencsg`?

I'm experiencing all these issues while creating threads, which I believe will be the most complex solid I'll be creating... that's why I started with them :roller_coaster:
So once I'm past this, hopefully my findings will benefit others making similarly complex solids.

**Test Cases**
Perhaps a good way to catalogue the issues I'm finding is by creating test-cases that fail (for the moment).
With an expectation that a `.clean()` will pass the test once I've figured out what's wrong with them.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Monday Oct 02, 2017 at 16:10 GMT_

----

@fragmuffin Threads are definitely the toughest challenge. Even with a high quality kernel, it is very slow and hard to make perform well.  I ended up doing a lot of odd tricks to get it to perform well when i did it in Onshape ( parasolid based).

I would hope that we can use the same cadquery version for any backend, but with less supported features. For example, there are still edges, faces, and solids, and the CQ 2.0 stuff we're talking about is still the same.  But I dont know what CQ features can be supported by a CSG backend. 

We removed the idea of both a FreeCAD and OCC backend, in CQ 2.0, having decided that the layer of abstraction was not worth it for this small of a difference.  I still agree with this decision-- having a different impl for CSG based kernel would be worth the cost of the abstraction layer if we can pull it off.

If it turns out that a CSG-based backend cannot support even the most basic CQ thigns, like ability to do workplanes, faces, and edges-- then it might make sense to make a whole separate project, as you have described.

adam-urbanczyk, 2019-06-13:
I will close this. For the time being there is a `fix()` function in the occ_impl layer now. Strategic plan is to move to OCCT7.3+ which should be more stable.

closed 2019-06-13 by adam-urbanczyk