CadQuery/cadquery #37: 2d Spiral primative

dcowden, 2018-12-06:
[image] **Issue by [fragmuffin](https://github.com/fragmuffin)**
_Tuesday Sep 19, 2017 at 11:04 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/196_

----

In the interest of #195, we also need a method of making a spiral primative

something like:

```python
import cadquery
from Helpers import show
part = cadquery.Workplane("XY") \
    .spiral(radius=1, growth=2, radius=108).close() \
    .extrude(3)
show(part)
```

looking into https://freecadweb.org/wiki/Part_Spiral/en now

dcowden, 2018-12-06:
[image] **Comment by [fragmuffin](https://github.com/fragmuffin)**
_Wednesday Sep 20, 2017 at 12:59 GMT_

----

FreeCAD's spiral primitive implementation:
[`App::DocumentObjectExecReturn *Spiral::execute(void)`](https://github.com/FreeCAD/FreeCAD/blob/949c86d364f05eecb5e7c279d6b2bc20e76e65ec/src/Mod/Part/App/PrimitiveFeature.cpp#L907)

It makes a wire along a conical surface, then projects it onto a plane.
And there doesn't seem to be an existing python hook for it.

adam-urbanczyk, 2019-06-13:
This is currently possible with parametric curves, cf. #110

closed 2019-06-13 by adam-urbanczyk