CadQuery/cadquery #39: Sweep Buggy for Sharp Paths (polylines)

dcowden, 2018-12-06:
[image] **Issue by [shivams](https://github.com/shivams)**
_Saturday Jul 08, 2017 at 02:51 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/186_

----

If the sweep path has sharp edges (e.g. sweep path is a poly-line), then sweep behaves in a buggy manner, and the object that is being swept doesn't turn correctly with the sweep path. 

In the following example, you can see that the "polyline" sweep is very thin after the "turn".

```
import cadquery as cq
from Helpers import show

pts = [(0,1), (0,20), (10,25)]

poly_path = cq.Workplane("YZ").polyline(pts)
spline_path = cq.Workplane("YZ").spline(pts)

sweep_poly = cq.Workplane("XY").circle(0.3).sweep(poly_path)
sweep_spline = cq.Workplane("XY").transformed(offset=(0, -3, )).circle(0.3).sweep(spline_path)

show(sweep_poly, (204, 100, 50, 0.2))
show(sweep_spline, (50, 204, 80, 0.2))
```

[image]

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Saturday Jul 08, 2017 at 02:59 GMT_

----

Hi @shivams 

What version of FreeCAD are you running, and on which operating system?

dcowden, 2018-12-06:
[image] **Comment by [shivams](https://github.com/shivams)**
_Saturday Jul 08, 2017 at 05:24 GMT_

----

Hi @jmwright 

I'm running FreeCAD 0.16 on Ubuntu 16.04. 

Doing sweep manually using the "Parts" workbench works fine in FreeCAD, though.

dcowden, 2018-12-06:
> An example:

[image] **Comment by [dcowden](https://github.com/dcowden)**
_Saturday Jul 08, 2017 at 11:48 GMT_

----

Thanks for reporting this. Can you help me understand more specifically
what is wrong in the example?

dcowden, 2018-12-06:
[image] **Comment by [shivams](https://github.com/shivams)**
_Saturday Jul 08, 2017 at 13:17 GMT_

----

Hi @dcowden 

As you can see in the previously attached image, the sweep done using the "polyline" path becomes very "thin" after the "turn". 

I have attached a new image, showing the difference in "sweep" done natively (manually) using FreeCAD (left panel in image) and that done using CadQuery (right panel in image). The native sweep done using FreeCAD is the desired behavior.

[image]

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Saturday Jul 08, 2017 at 14:11 GMT_

----

Oh I see yes thanks for that picture, it is much clearer now

So freedcad's implementation is  sweeping each segment with the profile
normal to each individual segment, but cq is not re-orienting the profile
after the first segment.

Can you please file an issue for this in github? I don't have time to look
now, but hopefully we can get to it. Of if you are adventurous, we are very
open to accepting pull requests.  I never seem to have a free hours but I
always have 5 minutes to point folks in the right direction...

The relevant code Is in the sweep method here:

https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/shapes.py

Looks like we call the freecad makePipe Shell method on Part.Wire...
perhaps there are some options that control the profile orientation we
could add:

https://www.freecadweb.org/wiki/TopoShape_API

Freedcad docs are not very good, they generally do not include the
overloaded signatures, so I usually have to use the python dir function at
the command line to figure out what is available...

Thanks for trying out cq!

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Saturday Jul 08, 2017 at 15:44 GMT_

----

These's a frenet option of FreeCAD's sweep implementation. It controls whether or not the face is reoriented during the sweep.

@shivams Please try the isFrenet option of our sweep operation.

https://github.com/dcowden/cadquery/blob/90acdbc440bd63347dfc546845f3fcf5a4ece3e1/tests/TestCadQuery.py#L376

dcowden, 2018-12-06:
[image] **Comment by [shivams](https://github.com/shivams)**
_Saturday Jul 08, 2017 at 18:46 GMT_

----

@jmwright:
Thanks for pointing out the `isFrenet` option. I had tried it and it produced the same "buggy" result. I should have mentioned this in my original post.

@dcowden:
Regarding opening an issue, isn't this already an opened issue? Or do I need to open another issue? 

And thanks for the links. I am busy for the next few weeks, however I'd really love to help when I will get time after that :+1:

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Monday Jul 10, 2017 at 02:25 GMT_

----

@shivams I think this issue is fine, you don't need to open another one. Can you post the code that you used to create the sweep manually in FreeCAD? CadQuery uses makePipeShell to perform the sweep operation.

https://github.com/dcowden/cadquery/blob/7b49cf6cb8c21ac676597b8fdd2c562a0335b43b/cadquery/freecad_impl/shapes.py#L941

dcowden, 2018-12-06:
[image] **Comment by [shivams](https://github.com/shivams)**
_Tuesday Jul 11, 2017 at 13:18 GMT_

----

@jmwright:
I created the model in FreeCAD using its visual interface. However, I have recorded a macro to achieve the same. You can run this macro and it will create that same sweep model. 

```
import FreeCAD
import Sketcher
import Part

App.newDocument("Unnamed")
#App.setActiveDocument("Unnamed")
#App.ActiveDocument=App.getDocument("Unnamed")
#Gui.ActiveDocument=Gui.getDocument("Unnamed")
App.activeDocument().addObject('Sketcher::SketchObject','Sketch')
App.activeDocument().Sketch.Placement = App.Placement(App.Vector(0.000000,0.000000,0.000000),App.Rotation(0.000000,0.000000,0.000000,1.000000))
#Gui.activeDocument().activeView().setCamera('#Inventor V2.1 ascii \n OrthographicCamera {\n viewportMapping ADJUST_CAMERA \n position 0 0 87 \n orientation 0 0 1  0 \n nearDistance -112.88701 \n farDistance 287.28702 \n aspectRatio 1 \n focalDistance 87 \n height 143.52005 }')
#Gui.activeDocument().setEdit('Sketch')
App.ActiveDocument.Sketch.addGeometry(Part.Line(App.Vector(0.000000,0.000000,0),App.Vector(0.000000,84.000000,0)),False)
App.ActiveDocument.Sketch.addConstraint(Sketcher.Constraint('Coincident',-1,1,0,1)) 
App.ActiveDocument.recompute()
App.ActiveDocument.Sketch.addConstraint(Sketcher.Constraint('PointOnObject',0,2,-2)) 
App.ActiveDocument.Sketch.addConstraint(Sketcher.Constraint('Vertical',0)) 
App.ActiveDocument.recompute()
App.ActiveDocument.Sketch.addGeometry(Part.Line(App.Vector(0.000000,84.000000,0),App.Vector(45.335453,108.429489,0)),False)
App.ActiveDocument.Sketch.addConstraint(Sketcher.Constraint('Coincident',0,2,1,1)) 
App.ActiveDocument.recompute()
App.ActiveDocument.recompute()
App.ActiveDocument.Sketch.delConstraint(2)
App.ActiveDocument.recompute()
#Gui.getDocument('Unnamed').resetEdit()
App.getDocument('Unnamed').recompute()
App.activeDocument().addObject('Sketcher::SketchObject','Sketch001')
App.activeDocument().Sketch001.Placement = App.Placement(App.Vector(0.000000,0.000000,0.000000),App.Rotation(-0.707107,0.000000,0.000000,-0.707107))
#Gui.activeDocument().activeView().setCamera('#Inventor V2.1 ascii \n OrthographicCamera {\n viewportMapping ADJUST_CAMERA\n  position 0 -87 0 \n  orientation -1 0 0  4.712389\n  nearDistance -112.88701\n  farDistance 287.28702\n  aspectRatio 1\n  focalDistance 87\n  height 143.52005\n\n}')
#Gui.activeDocument().setEdit('Sketch001')
App.ActiveDocument.Sketch001.addGeometry(Part.Circle(App.Vector(-0.000000,-0.000000,0),App.Vector(0,0,1),4.969474),False)
App.ActiveDocument.Sketch001.addConstraint(Sketcher.Constraint('Coincident',0,3,-1,1)) 
App.ActiveDocument.recompute()
#Gui.getDocument('Unnamed').resetEdit()
App.getDocument('Unnamed').recompute()
#Gui.activateWorkbench("PartWorkbench")
from FreeCAD import Base
App.getDocument('Unnamed').addObject('Part::Sweep','Sweep')
App.getDocument('Unnamed').ActiveObject.Sections=[App.getDocument('Unnamed').Sketch001, ]
App.getDocument('Unnamed').ActiveObject.Spine=(App.ActiveDocument.Sketch,["Edge1","Edge2"])
App.getDocument('Unnamed').ActiveObject.Solid=True
App.getDocument('Unnamed').ActiveObject.Frenet=True

```

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Tuesday Jul 11, 2017 at 13:23 GMT_

----

@shivams Thanks. I'm guessing there's a critical difference between using Part::Sweep and makePipeShell() that explains the glitch you're seeing. The macro you recorded should help figure out what the difference is.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Tuesday Jul 11, 2017 at 13:40 GMT_

----

Maybe we can just switch to using sweep. I don't think I elimiated using
sweep, so it means either I didn't know about it or it's been added since
freecad v11 (when I initially wrote cq)

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Tuesday Jul 11, 2017 at 13:42 GMT_

----

I actually implemented the sweep operation a year or so ago if I remember right. I don't remember Sweep being available then, but maybe I just missed it.

dcowden, 2018-12-06:
[image] **Comment by [fragmuffin](https://github.com/fragmuffin)**
_Wednesday Mar 14, 2018 at 09:44 GMT_

----

also, this duplicate function has to go

https://github.com/dcowden/cadquery/blob/7b49cf6cb8c21ac676597b8fdd2c562a0335b43b/cadquery/freecad_impl/shapes.py#L745

* [x] tick when done (PR #241)

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Wednesday Mar 14, 2018 at 12:43 GMT_

----

@fragmuffin What is that function a duplicate of?

dcowden, 2018-12-06:
[image] **Comment by [fragmuffin](https://github.com/fragmuffin)**
_Wednesday Mar 14, 2018 at 23:31 GMT_

----

@jmwright
`Solid.sweep` is declared twice, the second declaration replaces the first.
Much like writing `x=1; x=2`, the `x=1` is functionally ignored.

dcowden, 2018-12-06:
[image] **Comment by [fragmuffin](https://github.com/fragmuffin)**
_Thursday Mar 15, 2018 at 13:04 GMT_

----

@jmwright PR #241 didn't fix this issue, I just removed part of the code that had no function.

I started to look into this bug, but I haven't been able to find anything useful to fix it.

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Thursday Mar 15, 2018 at 14:54 GMT_

----

@fragmuffin Understood. Are you working on fixing this issue, or was #241 a stand-alone submission?

dcowden, 2018-12-06:
[image] **Comment by [adam-urbanczyk](https://github.com/adam-urbanczyk)**
_Thursday Mar 15, 2018 at 21:22 GMT_

----

Maybe this will help: http://opencascade.blogspot.nl/2009/12/surface-modeling-part4.html (but you'd need to browse thorough FreeCAD code)

Once I have some time I will look myself too.

dcowden, 2018-12-06:
[image] **Comment by [fragmuffin](https://github.com/fragmuffin)**
_Thursday Mar 15, 2018 at 23:17 GMT_

----

#241 was a stand alone. I don't think I'll have time to fix this one.

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Tuesday Apr 10, 2018 at 13:09 GMT_

----

I accidentally closed this 26 days ago. We haven't addressed the original issue so I'm reopening.

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Tuesday Apr 10, 2018 at 13:57 GMT_

----

It looks like the Part workbench does quite a bit of work before it calls makePipeShell. https://github.com/FreeCAD/FreeCAD/blob/a07226cea111ddf77c3a416551f0227011efd0c5/src/Mod/Part/App/PartFeatures.cpp#L358

In our implementation we just [call makePipeShell directly](https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/shapes.py#L943). The naive approach that I took when I implemented the sweep operation is probably where the bug with sweeping over the sharp polyline path comes in.

The Part workbench implementation assumes that it has access to GUI-level classes, which in CadQuery we don't. It's no small task to distill the important checks being done and then re-implement them in our Solid class. I think that's what needs to be done to fix this though.

@shivams Unfortunately, I'm not sure when this will be fixed. If you're still monitoring this issue, you could check out a sweep operation feature that was just added to see if that helps in the meantime. https://github.com/dcowden/cadquery/blob/master/examples/FreeCAD/Ex034_Sweep_Along_List_Of_Wires.py

adam-urbanczyk, 2019-06-07:
Solved by #112

closed 2019-06-07 by adam-urbanczyk