CadQuery/cadquery #41: Define Gui direction for CQ 2.0

dcowden, 2018-12-06:
[image] **Issue by [dcowden](https://github.com/dcowden)**
_Sunday Dec 11, 2016 at 21:24 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/171_

----

What GUI will we use, now that we will lose FreeCAD underpinning?

   * wxPython based ( powered by pythonOCC scaffolding )
   * electron based gui?

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Wednesday Dec 14, 2016 at 02:42 GMT_

----

I think there are a few ways we can go. I did some work on an [Electron GUI](https://github.com/jmwright/cadquery-gui) that was decoupled from the CadQuery backend by passing JSON for the renders. The Electron app was coming together pretty well with one big drawback - CPU utilization. That's a pretty common problem with Electron I guess.There are the PythonOCC GUI examples, and I think I used one of those to build the [cadquery-x](https://github.com/jmwright/cadquery-x) experimental interface. I had actually taken that GUI off of GitHub and someone tracked me down through the FreeCAD forums and asked me to put it back up. They were using it as an example.

I don't know that I have a strong preference on the GUI other than I would like for it to be decoupled from the backend so that it could be used with more than just the CQ backend (NodeOCC, PythonOCC, FreeCAD, etc). Any CAD library that can take a script in and give a JSON representation of the 3D model would be fair game. 

The main choices that I see for GUI development are these:
1. PySide (Qt implementation without the licensing entanglements)
2. wxPython
3. Electron

One of the main advantages of Electron is that it can run as a desktop app, but the code can be written in such a way that it will also run as a web app through a browser if desired. There will likely be performance challenges with that method though. It's apparently possible to overcome those challenges as web-based CAD apps like Onshape show, but with our resources there's a possibility that we might hit a wall at some point. Would taking that chance be worth it for what Electron gives us in return? I don't know.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Wednesday Dec 14, 2016 at 03:05 GMT_

----

> I don't know that I have a strong preference on the GUI other than I would like for it to be decoupled from the backend so that it could be used with more than just the CQ backend (NodeOCC, [...]

I agree with you completely.  I'd be all-in on this approach if I could figure out one thing: how do you go about 'selecting things' when the gui is decoupled from the backend? I think the performance problems are solvable, but I'm not sure offhand how to allow, for example, selecting an edge in a way that the backend can 'know' which edge was selected.  Most of the display strategies seem to use a mesh approach-- so the selection ends up being hard to correlate back to the BREP on the backend.

Do you have ideas on how this would work?

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Wednesday Dec 14, 2016 at 03:09 GMT_

----

I've seen feature selection examples using raycasting, but I'm not sure how you relate that information to the back end. Maybe if the JSON that's sent to the GUI has the feature IDs embedded, that might allow us to correlate the BREP to the render that's being displayed to the user.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Wednesday Dec 14, 2016 at 03:20 GMT_

----

hmm that could work. I suppose also its possible to use non-mesh representations that three.js offers.

There's also a possible overlap with another issue i'm mentally struggling with. Bear with me on this, it's difficult to explain quickly.

One of the things we desperately need in CQ2 is a 'created by' selector, so that you can filter selections to only those 'created by' a given operation.   Taking an example, consider a cube with a hole drilled into one face.   The operation that creates the hole creates two faces ( the bottom of the hold and the cylinder that forms its side), while modifying one face of the cube.  If the operation that drilled the hole was called 'drill_hole", we'd like to be able to do something like 

>  .created_by('drill_hole').faces()

After investigating , I've concluded that OCC provides absolutely no support for this at all.  What I think we'll have to do is compute some kind of 'fingerprint' for each face, edge, and solid, and then compare the result of an operation with that original set.  Anything that has changed from the previous is recorded as 'changed' by the operation.   As a not-really-working-but-illustrates-the-point  idea, you could create an MD5 has of a stream that consists of object type + xyz coordinates sampled along the object, and then use that hash for identity.

Phew! anyway, that all relates to your comment only in the way that each edge, face, and object would already _have_ a featureID we could use for selection....

dcowden, 2018-12-06:
[image] **Comment by [adam-urbanczyk](https://github.com/adam-urbanczyk)**
_Thursday Dec 29, 2016 at 19:09 GMT_

----

+1 for PyQt/PySide and reusing interactive features of OCC

I did some work in this direction: https://bitbucket.org/79labs/cadquery-testbench/

dcowden, 2018-12-06:
> cadquery-testbench/

[image] **Comment by [dcowden](https://github.com/dcowden)**
_Thursday Dec 29, 2016 at 21:32 GMT_

----

Ok, that's a plan then!

Adam your work looks great! It's actually closer than I thought to
something functional. I've always been wary of trying to build a competitor
to freedcad ( IE, another free cad package), but maybe it's doable if we
keep to the code CAS route and avoid getting trapped into build a point and
click cad

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Friday Feb 23, 2018 at 14:43 GMT_

----

[cadquery-gui](https://github.com/jmwright/cadquery-gui) is being actively developed again. It's in a good position to support the PythonOCC version of CadQuery eventually. It uses the FreeCAD version for the backend now. The main problem with it before was high CPU utilization due to the 3D view. That has been solved now using static rendering of the 3D scene rather than dynamic rendering.

It's not ready for wide-spread use yet, and there's not really any documentation on how to use it, but if anyone wants to experiment with it you can run any CQGI-compliant script (calls show_object). cadquery-gui (cqg) doesn't try to recreate the great functionality that's in everyone's favorite editors. It relies on an external editor (like Atom, VIM, Emacs, Sublime, etc). If you open a script using the file menu it should render, and you can click the open button (right next to the Execute/Build button) to open the script in an external editor. You can change what command is run for the editor in the Settings dialog. Every time you save the script cqg will execute it and update the 3D view, and there's a setting to turn that off too. Ctrl+Shift+C will open the debugging console so that you can see if the script (or GUI) is throwing errors.

Please keep in mind that it's still early in the development process, but I've started using it instead of CadQuery's FreeCAD module. Things like built-in documentation and better script debugging are planned next.

jmwright, 2019-12-16:
I feel like this issue has taken care of itself, and I will list the top options below as I currently understand them. Other community members should feel free to comment with dissenting opinions and reopen this issue if they feel it's warranted.
* [CQ-editor](https://github.com/CadQuery/CQ-editor) is the primary desktop GUI development environment for CQ 2.0, and is based on Qt. We're still struggling with some issues on MacOS though, which may make cadquery-gui (listed below) a good alternative for those users. CQ-editor provides advanced features including debugging and a variable inspector.
* [jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery) is the primary web environment for CQ 2.0 scripts, and is available on [MyBinder.org](https://mybinder.org/v2/gh/bernhard-42/jupyter-cadquery/master?urlpath=lab&filepath=examples%2Fcadquery.ipynb). It's a great way to work with CadQuery in the cloud. It also makes it easy to integrate lab notes and equations with your CadQuery scripts.
* [cadquery-gui](https://github.com/jmwright/cadquery-gui) is a simple viewer that is compatible with CQ 1.0 and CQ 2.0. It is Electron-based and seems to work well on Windows, Linux and MacOS. This can be helpful for MacOS users who are struggling to get CQ-editor running (we're still working to address those bugs). It relies on an external text editor (VS Code, PyCharm, etc), there is no integrated code editor like there is with CQ-editor. I feel that this fills a niche and is a worthwhile inclusion, but it's also a project of mine, so maybe I'm biased.

There are other options in the community, but I feel like these are the three maintained options that are capable of running CQ 2.0 scripts today. Let me know if there are others I've missed.

closed 2019-12-16 by jmwright