CadQuery/cadquery #42: Implement Parametric 2d sketching

dcowden, 2018-12-06:
[image] **Issue by [dcowden](https://github.com/dcowden)**
_Sunday Dec 11, 2016 at 21:59 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/172_

----

Currently, sketching is not parametric.  Sketches explicitly define geometry, rather than the design intent. 

For example, instead of defining a square as four lines between points 0,10, 10,10, 10,0, and 0,10, we should define four lines with these constraints:
    each of the ends co-incident
    length of each line equal
   one line equal to 10
   one point co-incident with the origin

To do this, we will need to integrate a constraint solver.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Sunday Dec 11, 2016 at 22:03 GMT_

----

This is the solution for the functionality proposed in #155.

dcowden, 2018-12-06:
[image] **Comment by [adam-urbanczyk](https://github.com/adam-urbanczyk)**
_Thursday Dec 29, 2016 at 19:44 GMT_

----

2D constraint solver implemented in js : https://github.com/xibyte/jsketcher

dcowden, 2018-12-06:
> 2D constraint solver implemented in js : https://github.com/xibyte/
> jsketcher

[image] **Comment by [dcowden](https://github.com/dcowden)**
_Thursday Dec 29, 2016 at 21:37 GMT_

----

Too bad it's agpl, we can't use it.

That said, I really thing cq 2.0 needs to contemplate parametric 2d
sketching, even if we implement it later on. I'll at least include an idea
of how it might work in the 2_0 branch

darrellenns, 2020-10-01:
Perhaps something like Z3 could be leveraged for solving constraints?

jmwright, 2020-10-01:
Thanks @darrellenns for the suggestion.

Link: https://github.com/Z3Prover/z3

Python bindings: `pip install z3-solver`

adam-urbanczyk, 2020-10-02:
@darrellenns would you know if it is better then using numerical optimization methods (Newton or BFGS)?

darrellenns, 2020-10-03:
@adam-urbanczyk I'm not entire certain. If the numerical methods are sufficient, I would guess that they may perform better. However, this is a bit outside my area of expertise. I just know that Z3 is powerful and popular constraint solver, but I'll leave it to others to decide if it's well-suited to this particular application.

MarcWeber, 2020-11-03:
https://solvespace.com comes with C code. I haven't checked license.

adam-urbanczyk, 2020-12-23:
Solvespace is GPL so out of question. The FreeCAD solver (FreeGCS) could be an option once extracted and wrapped via pybind11.

PR #879 (2021-11-11): Sketch support
- [x] Tests
- [x] Integration with cq.Workplane
- [x] Docs

Will resolve #42 #52
closed 2021-11-11 by marcus7070