CadQuery/cadquery #48: Implement Text Generation

dcowden, 2018-12-06:
[image] **Issue by [dcowden](https://github.com/dcowden)**
_Sunday Dec 11, 2016 at 21:05 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/166_

----

Find a way to generate text.

The ideal implementation requires a baseline and a face/surface-- the text should follow the surface.

adam-urbanczyk, 2018-12-07:
Seems that PythonOCC supports this: https://github.com/tpaviot/pythonocc-demos/blob/02ad16678a01f2047d110331f77c8ceabd34bfb3/examples/core_font_helloworld.py

dcowden, 2018-12-07:
ah! that's awesome! I looked into it in FreeCAD and it seemed hard, but this seems pretty easy to integrate. Yay for getting rid of the FreeCAD layer!

PR #106 (2019-02-22): Support for text rendering 
This will resolve #48 

[image]
closed 2019-02-22 by adam-urbanczyk