CadQuery/cadquery #50: Implement Feature Tree

dcowden, 2018-12-06:
[image] **Issue by [dcowden](https://github.com/dcowden)**
_Sunday Dec 11, 2016 at 21:02 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/163_

----

CQ 2.0 should have a feature tree, which stores operations and their results in a way that can be searched and referenced by later steps.

Each branch in the tree should have an id, and an associated operation.

The tree will be stored in the modelling context.

marcus7070, 2020-05-31:
This is being talked about a lot in #371, so I thought I'd write down some ideas here.

I think one of the main goals of a feature tree implementation should be to enable selection by operation, such as 
```python
filleted = cbore.faces( 
      OperationSelector(createdBy=”cbore”) && NormalSelector(“Z”) )
      .edges().chamfer(0.1 )
```
from [here](https://docs.google.com/document/d/1CwVgozH1Amx4lcIhvJfWMpFQ9qjYG8rBb0l75czL4wE/edit?usp=sharing).

I would define an operation as a method that returns a new or modified `Solid` or `Compound` in `CQ.objects`.

So for each "operation" we have to store a `cadquery.occ_impl.Shape` object (or a list of them) representing the operation. Subsequent selectors like `NormalSelector` in the example above then act on this `Shape`. 

If that's the way to achieve `OperationSelector`, then I guess the most obvious path is to modify methods we class as operations to log an operation `Shape`, such as
* `union`
* `cut`
* `cutEach`
* `_combineWithBase`
* `_cutFromBase`

The good bit here is that lots of methods like `extrude` call `_combineWithBase`, so we can catch a heap of operations in `_combineWithBase`. And some of the methods that directly use `Shape.fuse` to combine `Shape` objects should probably be rewritten to all use a common method like `_combineWithBase`. 

More complicated operations such as:
* `fillet`
* `chamfer`

would have to implement their own logic to create the operation `Shape`, probably involving a difference between the end and starting solid.

Transformations such as offsets and rotations would have to be logged and replayed on the operation `Shape` when it is fetched.

---

Am I missing any other method to achieve `OperationSelector`? I find it difficult to read the OCCT documentation, I could easily be missing something useful in there.

I did think about doing some fancy stuff with `CQ.__getattribute__` to log each method and it's arguments used and then recreate the operation, but that just seems too complicated and fragile compared to the above method.

adam-urbanczyk, 2020-05-31:
Note that essentially all OCCT operation implement `Modified` and `Generated`: https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_b_rep_builder_a_p_i___make_shape.html . That includes fillets and chamfers. We should use the kernel as much as possible.

Another question is how will it actually fit in the current framework. For example: what should happen if the selected object is not part of the current solid anymore? Should it be then selected wit`forConstruction=True` or not selected at all?

marcus7070, 2020-05-31:
Thanks @adam-urbanczyk! 

I also found this introduction to OCCT's history support which really helped me understand: https://dev.opencascade.org/doc/overview/html/occt_user_guides__modeling_algos.html#occt_modalg_hist

I need to read up more about these OCCT functions.

dcowden, 2020-05-31:
@marcus7070  @adam-urbanczyk Good question about what happens when objects that were there are not any more.

I played around with the Modified and Generated methods at one point, and i remember being disappointed, and learning that the behavior i'd like is much harder that it first appears.  IIRC ( and if it's not changed since), most methods ( for example, cutting a hole in a face) return a newly generated face, not the 'original'

Then, it becomes more obvious that 'original' isn't really even well defined.   If I select the top face of a box, and i simply cut a hole, i would normally expect that the face is the same as before-- but why? if you split a face into two faces, are they both the 'original'? or are they both new?  What does 'new' really mean? 

I use  Onshape and Solidworks a lot, and what I found is that they tend to define selections as the model evolves by tracking the history, and being fairly broad about how much an object can change and still be called 'the same face' for selection purposes. IE, if a face is created in step 1, and selected, all faces that used to be part of that face are counted as the original face.  It's definitely clear that once a face is no longer in the same plane as it originally was, its not 'the same'. Solidworks does much better than Onshape, with better being defined VERY nebulously as 'what i expect'.  

Even so, it is pretty frequent that a face/edge/point changes such that later selectors fail. one example is if a vertex disappears. another is that the face is no longer defined.  In these cases, the model will rebuild, but the features that use the deleted reference will be suppressed, or a 'guess' will be used.

To explain a 'guess', I need to digress a bit about 2d sketching.   In Onshape, FreeCAD, SW, 2d sketches are parametric too-- so you can define constraints and dimensions, not just develop with the 2d vertices.  When you sketch objects initially, they are drawn on the screen, using in the screen's scale to record the x,y locations of the vertices.  Since these are not dimensioned or constrained, they can all be moved around-- but the values are still stored as 'guesses'-- so that i you exit the sketch, the entities will appear as you left. them.  Indeed, you can even work with this sketch, including extruding it. 

When you add sketch constraints, the solver will try to solve the sketch. The solver is allowed to move points around to meet the constraints, until all of the points are fully constrained.  At this point, and changes  will re-solve the sketch, and potentially move the points.

In Solidworks and Onshape, if the constraints of a sketch are based on projected geometry, and that geometry becomes unavailable, the constraint will be disabled. In the GUI, you'll usually see a 'deleted' or similar, telling you it used to be there, but it's gone now.  The sketch will rely on the 'guesses' from when it was initially created when it is rendered.  The overall model will attempt to render with the constraint disabled. Sometimes, it's just fine, because the sketch, while underconstrained, works just fine with the original guesses.  Other times, you'll get very odd results.

dcowden, 2020-05-31:
My current thinking along the lines of the 'feature tree'  looks something like the below.

CQ layers above the OCP layer should assume that they may be used directly, in addition to used by the fluent api.  

the CQ fluent api currently acts as a feature tree, but it is usable outside the fluent api.  (As an example, it should be possible to build up a model using operations and selectors directly, but still benefit from ability to query the tree/history of objects.

[image]

marcus7070, 2020-06-01:
> it is pretty frequent that a face/edge/point changes such that later selectors fail

I used Solid Edge/Solidworks for work from 2002 to 2014, and this was one of the biggest problems in a long list of problems. It was so unreliable and un-debuggable that I wound up ignoring most of the sketching features in Solidworks and doing as much as possible in Excel, then just the minimum final few steps in Solidworks. This insane workflow about doubled my productivity with complex models. 

So I would encourage CadQuery to do the minimum required of these fuzzy relationships.

marcus7070, 2020-06-01:
> it should be possible to build up a model using operations and selectors directly, but still benefit from ability to query the tree/history of objects

That sounds good. 

So most of the examples seem to involve a function call like `part.revolve(270, (0, 0, 0), (0, 0, 1), opName="foo")`. The naive solution would be to pass those opNames down to the Core layer, so it can identify the feature and store it in a feature tree. I wonder if we can function more like how tags currently work: `part.revolve(270, (0, 0, 0), (0, 0, 1).tagOperation("foo")`. Otherwise we are going to have to repeat the `opName=None` in hundreds of function definitions and calls.

dcowden, 2020-06-01:
@marcus7070 

> So I would encourage CadQuery to do the minimum required of these fuzzy relationships.

I agree.  Let's try to get as much of the functionality we can, without crossing the line into unpredictable behavior.  You're not a real Solidworks user unless you have had a 90% complete model get hopelessly turned to red errors and had to start completely over :)

> Otherwise we are going to have to repeat the opName=None in hundreds of function definitions and calls.

Excellent point!.  maybe a way to think about this would be to imagine how it works when interacting in lower level apis directly (  selector api, the operations api, and the feature tree ) , and then after that, rolling it up into the fluent api? I think we'll probably find that your proposal is better, because in 3gl mode we'll be looking at a distinct operation call too. Just experimenting with a 'lower level' api below, I can already see that its hard to avoid involving each operation with the feature tree ( each operation has ctx=ctx):

```
#
# the ever-loved block-with-a-hole, imagined 
# using a 3gl program layer
import cq

#
# context is the feature tree
#
ctx = cq.Context()
wp = cq.Workplane(plane='XY', ctx=ctx)

sk = cq.Sketch(wp,ctx=ctx)
sk.rect((0,0),(20,20))

#in the low level api, operations give direct access to "Generated and Modified",
#as well as ability to trap exceptions and deal with conditional results

extrude_op = cq.op.Extrude(sketch=sk,ctx=ctx,name="make_base_boss",raise_on_failure=True)
extrude_op.execute()

startFace = DirectionSelector(">Z") && OperationSelector(createdBy="make_base_boss") && TypeSelector("face")
upToFace = DirectionSelector("<Z") && OperationSelector(createdBy="make_base_boss") && TypeSelector("face")

wp = cq.Workplane(plane=s, ctx=ctx)

sk2 = cq.Sketch(wp,ctx=ctx)
sk2.circle(0.5)

cut_op = cq.op.Cut(sketch=sk2,upTo = upToFace, ctx=ctx , name="the hole")
```
Factory methods on the tree would stink-- then we'd have to have a functional copy of every object. 
Context managers don't seem like the best solution here either:
``` 
with cq.Context() as ctx:  
     #code still has to inject ctx manually

adam-urbanczyk, 2020-06-01:
OK, after some thinking my personal conclusion is:

1) Implementing a full-blown feature tree is rather involved and looks like it is not worth the effort. It is not clear to me if it makes sense at all in code based CAD tools.
2) It would be good to have the possibility to tag operations and *maybe* query their modified/created shapes. I'd propose to stick to the `tag` terminology (thus not `name`) to make things consistent and just extend existing capabilities somewhat.
3) We definitely need a separate `Sketch` class that can reference items down the stack (also indicated in this issues: #361, #52).  
4) Nice idea with the picture @dcowden - I'll have some additions but we definitely need something like that to stay focused.

dcowden, 2020-06-01:
@adam-urbanczyk looking forward to your additions/ suggestions. I agree with sticking with 'tag'

Another idea that might be  worth considering: perhaps we should create an established set of 3d objects that form the basis for scenarios we want to consider.  It's sometimes very hard to think about all of the scenarios up-front from the air: it would be nice to have a concrete list of objects so that one can say "ah, yes, that makes scenario 1 easy, but it's not at all clear what that would do to scenario 7"

This set of objects would also serve as very useful for testing: someone familiar with the project should very quickly and immediately know whether they've introduced a regression/problem.   It's possible to automatically check that, for example, by using the mass as a checksum