CadQuery/cadquery #52: The idea of adding a sketchlet function into the 2D operations

dcowden, 2018-12-06:
[image] **Issue by [huskier](https://github.com/huskier)**
_Friday Nov 18, 2016 at 05:59 GMT_
_Originally opened as https://github.com/dcowden/cadquery/issues/155_

----

The idea is that we could store some useful sketches in data files( maybe a special formatted file), and we could re-use them. The sketchlet function could parse the special formatted files or just strings, and draw them.

With sketchlet function, we could significantly simplify the example "Panel With Various Connector Holes", and make the code elegant.

The following code is just a simple demo, no any parsing function. **The current problem is that we could only add one complete wire into the sketchlet wire, otherwise Wire.assembleEdges complaint that "command not done".** In the following code, I commented the second circle code.

`

    def sketchlet(self):
        def makeSketchletWire(pnt):
            p0 = pnt
            p1 = p0.add(Vector(-23.5,0,0))
            e1 = Edge.makeCircle(1.6,p1)
            #p2 = p0.add(Vector(23.5,0,0))
            #e2 = Edge.makeCircle(1.6,p2)
            listOfEdges = []
            listOfEdges.append(e1)
            #listOfEdges.append(e2)
            sketchWire = Wire.assembleEdges(listOfEdges)
            sketchWire.forConstruction = False
            return sketchWire       
        return self.eachpoint(makeSketchletWire, useLocalCoordinates=True)
`
To call sketchlet:
`

result = cadquery.Workplane("XY").pushPoints( [ (25, 25),(50, 50)] ).sketchlet().extrude(50)
show(result)

`

dcowden, 2018-12-06:
[image] **Comment by [huskier](https://github.com/huskier)**
_Friday Nov 18, 2016 at 13:50 GMT_

----

The problem is how do we know how many individual wires can be obtained from the listOfEdges?
`

```
def assembleEdges(cls, listOfEdges):
    fCEdges = [a.wrapped for a in listOfEdges]
    wa = Wire(FreeCADPart.Wire(fCEdges))
    return wa
```

`

`

 from FreeCAD import Base 
 e1 = Part.makeCircle(1.6, Base.Vector(-23.5,0,0), Base.Vector(0,0,1))
 e2 = Part.makeCircle(1.6, Base.Vector(23.5,0,0), Base.Vector(0,0,1))
 ws = Part.Wire([e1,e2])

`
The above code in FreeCAD give error that reads:

> Traceback (most recent call last):
>   File "<input>", line 1, in <module>
> Part.OCCError: BRep_API: command not done

It seems that this is an error from FreeCAD, not CQ.

I report an issue at FreeCAD's website, and the web link is as:
http://www.freecadweb.org/tracker/view.php?id=2777#bugnotes

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Saturday Nov 19, 2016 at 19:51 GMT_

----

Hi, @huskier 

I love the idea of a sketchlet.  Its closely related to another feature I've been wanting to implement-- truly parametric 2d sketches.

Tell me more about what data format/data you envision inside of a sketchlet.  If you have 'static' geometry in mind, i suspect that using DXF would be the best strategy. DXF import would essentially be a 2-d sketchlet, but would be much easier for users to create because so many DXFs already exist, and they can be exported by nearly any program.

I have used this package before with great success: https://pypi.python.org/pypi/dxfgrabber

I would envision something like this:

`result=cadquery.Workplane("XY").import("/path/to/file.dxf").extrude(50)`

Now, things get a bit more interesting if you imagine that sketchlets are parametric. For example, i dont simply have a square defined by  coordinates at (0,0), (10,0), (10,10), and (0,10) with lines inbetween ( as would be done with existing CQ).

 Instead, I have a square, which is composed of 4 lines, with:
    \* two lines having vertical constraints
    \* two lines have horizontal constraints
    \* all 4 lines having an equal length constraint (or alternatively, the vertices of each line endpoint having co-incident constraints)
    \* one side having a length constraint set to 10
- a co-incident constraint for one vertex of one line with the origin

Now _this_ is a different ball of wax-- because now the sketch itself can preserve design intent, while allowing the single value, 10, to be changed.

Implenting this would require using a geometry solver. But would be really cool!

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Saturday Nov 19, 2016 at 21:27 GMT_

----

@huskier ,@jmwright

should we be interested in doing parametric sketches, here's one candidate library for it:

http://geosolver.sourceforge.net/

Another possibility is to use the Cheetah solver, which is what FreeCAD uses ( read here)
http://www.slideshare.net/NickSidorenko/beyond-parametric-new-approach-to-geometric-constraint-solving

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Saturday Nov 19, 2016 at 21:41 GMT_

----

actually, maybe FreeCAD doesnt use Cheetah any more. Their solver can be found here. Maybe it is an older version of Cheetah, but since it is open source we could use it. Or we could write a wrapper around it since we already use FreeCAD

https://github.com/FreeCAD/FreeCAD/tree/master/src/Mod/Sketcher/App/planegcs

dcowden, 2018-12-06:
[image] **Comment by [jmwright](https://github.com/jmwright)**
_Sunday Nov 20, 2016 at 21:35 GMT_

----

@dcowden I don't have a problem with 2D parametric sketching being added to CadQuery.

dcowden, 2018-12-06:
[image] **Comment by [huskier](https://github.com/huskier)**
_Monday Nov 21, 2016 at 01:34 GMT_

----

Could we discuss this in two steps, first 'static' geometry, and then 'parametric' geometry? And for now, could we firstly focus on the 'static' geometry aspect?

`I think we can firstly make the following demo code work properly in CQ, and then talk about 'DXF' things.

And the latest progess is at http://www.freecadweb.org/tracker/view.php?id=2777. And we could do the multiple wires sketch in FreeCAD itself, but we could not do it in CQ at present. Like the following code, if we uncommented  #listOfEdges.append(e1) and  #listOfEdges.append(e2), the code could not run.

    def sketchlet(self):
        def makeSketchletWire(pnt):
            p0 = pnt
            p1 = p0.add(Vector(-23.5,0,0))
            e1 = Edge.makeCircle(1.6,p1)
            p2 = p0.add(Vector(23.5,0,0))
            e2 = Edge.makeCircle(1.6,p2)
            p3 = p0.add(Vector(-17.038896,-5.7,0))
            p4 = p0.add(Vector(-19.44306,-4.70416,0))
            p5 = p0.add(Vector(-20.438896,-2.3,0))
            e3 = Edge.makeThreePointArc(p3,p4,p5)
            p6 = p0.add(Vector(-21.25,2.3,0))
            e4 = Edge.makeLine(p5,p6)
            p7 = p0.add(Vector(-20.25416,4.70416,0))
            p8 = p0.add(Vector(-17.85,5.7,0))
            e5 = Edge.makeThreePointArc(p6,p7,p8)
            p9 = p0.add(Vector(17.85,5.7,0))
            e6 = Edge.makeLine(p8,p9)
            p10 = p0.add(Vector(20.25416,4.70416,0))
            p11 = p0.add(Vector(21.25,2.3,0))
            e7 = Edge.makeThreePointArc(p9,p10,p11)
            p12 = p0.add(Vector(20.438896,-2.3,0))
            e8 = Edge.makeLine(p11,p12)
            p13 = p0.add(Vector(19.44306,-4.70416,0))
            p14 = p0.add(Vector(17.038896,-5.7,0))
            e9 = Edge.makeThreePointArc(p12,p13,p14)
            e10 = Edge.makeLine(p14,p3)

	    listOfEdges = []
            #listOfEdges.append(e1)
            #listOfEdges.append(e2)
            listOfEdges.append(e3)
            listOfEdges.append(e4)
            listOfEdges.append(e5)
            listOfEdges.append(e6)
            listOfEdges.append(e7)
            listOfEdges.append(e8)
            listOfEdges.append(e9)
            listOfEdges.append(e10)

            sketchWire = Wire.assembleEdges(listOfEdges)
            sketchWire.forConstruction = False
            return sketchWire
        return self.eachpoint(makeSketchletWire, useLocalCoordinates=True)

`

dcowden, 2018-12-06:
[image] **Comment by [huskier](https://github.com/huskier)**
_Monday Nov 21, 2016 at 06:36 GMT_

----

I like python library dxfgrabber. If we add 'static' geometry import function with dxf support, we need to use dxfgrabber, and the question is whether we include dxfgrabber library as a part or just make it as a dependent library?

BTW, maybe we need to add another function to expand the arc-making method, since the current makeThreePointArc(...) sometimes inconvenient.

With DXF file importing function, we may make CQ having 3D text supporting, and I think it is a very exciting feature. Thinking the following workflow:
1. workplane("XY").text("CQ", "some styles......")
2. write to a dxf file
3. import the dxf file

OR, we could make 2 & 3 implicitly, and we just do:
workplane("XY").text("CQ", "some styles......").extrude(20)

And, then, we could print 3D TEXTs!

Too optimistic. 3D texts are not that easy!

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Monday Nov 21, 2016 at 11:45 GMT_

----

Yes, we have discussed this before and decided that we are willing to begin allowing dependencies.  There are already so many that it's a problem needing a solution anyway. We need to provide requirements.txt for easy installation.

There are two ways to make cq easier to install initially, including freecad.  One would be to provide a conda package, the other to provide a docker container. I have started and mostly completed that work a few months back, but for side tracked on some other projects.

When that work is completed, extra libraries will not be any harder for the user to install

dcowden, 2018-12-06:
[image] **Comment by [huskier](https://github.com/huskier)**
_Tuesday Nov 29, 2016 at 01:13 GMT_

----

Could we move into the next step? @dcowden & @jmwright  Could you help to make the following simple demo workable in the current master branch? Then we could do the 'static geometry' importing things.

`

    def sketchlet(self):
        def makeSketchletWire(pnt):
            p0 = pnt
            p1 = p0.add(Vector(-23.5,0,0))
            e1 = Edge.makeCircle(1.6,p1)
            p2 = p0.add(Vector(23.5,0,0))
            e2 = Edge.makeCircle(1.6,p2)
            listOfEdges = []
            listOfEdges.append(e1)
            listOfEdges.append(e2)
            sketchWire = Wire.assembleEdges(listOfEdges)
            sketchWire.forConstruction = False
            return sketchWire       
        return self.eachpoint(makeSketchletWire, useLocalCoordinates=True)
`

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Tuesday Nov 29, 2016 at 01:37 GMT_

----

@huskier i would rather work on just importing DXF next. 

I think we're going to run into problems quickly getting the example above to work nicely, due to the limits on the number of wires ( which you observed earlier ).

This might help: its the current 'best' way to link in new functions

http://parametricparts.com/docs/extending.html

dcowden, 2018-12-06:
[image] **Comment by [huskier](https://github.com/huskier)**
_Tuesday Nov 29, 2016 at 05:40 GMT_

----

@dcowden With the help of dxfgrabber, it is not so difficult to import DXF file. However, DXF drawings are usually more complex than CQ's native drawing. I am afraid we have to deal with the limits on the number of wires.

Another issue is that the dependencies. For the scripting from CLI, maybe it is not a big problem, and we just install the dependencies. However, for the FreeCAD's plugin, we have to think a way to make it work before the complete of the conda packages and the docker containers.

Anyway, we could just implement the function first.

Meanwhile, I will think about a solution to the limits on the number of wires.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Tuesday Nov 29, 2016 at 12:22 GMT_

----

@huskier yes, I agree there are two main audiences here: 

(1) people who are using FreeCAD, and are using the CQ plugin ( and are thus also probably using FreeCAD's GUI ), and 
(2) people who are using CQ as  a stand alone tool.

A docker image and conda package are clearly ideal for audience (2).  I think its actually possible that it is ideal for a portion of audience (1), but I agree it is not suitable for people in group (1) who have already installed FreeCAD before they come to use CQ. They do not want to repeat/remove their install.

I assume that FreeCAD has DXF import. Since our implementation is based on FreeCAD, i assume we can easily test the feasibility and wire limits by importing DXF from within FreeCAD. If it works there, we can make it work too.

I think importing DXF will be, in practice, much easier to code than the sketchlets you have proposed.  With sketchlets we need to have the syntax fit in nicely with the CQ code flow ( the chainability, etc).  If we have a single new function called dxf('filename'), the implementation of this function can natively call FreeCAD libraries, create wires, etc, with impunity.  I would envision it creating a multitude of wires ( representing loops in the underlying DXF), and adding them all to the stack. I dont see it being a problem.

dcowden, 2018-12-06:
[image] **Comment by [huskier](https://github.com/huskier)**
_Wednesday Nov 30, 2016 at 05:50 GMT_

----

OK, let's work on DXF importing.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Wednesday Nov 30, 2016 at 12:16 GMT_

----

Ok, I will start the process with:

  * a document describing how it should work
  * a branch with a start 

I have done some work already with dxfgrabber in the past.  Oddly enough, I also ended up using another dependency, networkx, for finding loops. And merging nodes that were subject to round-off error.  The main difference between a low and high quality DXF importer is whether it handles merging loops so that there are not geometry problems. Many DXFs are created with 'dumb' lines and arcs that do not meet exactly at the ends. 

I'll try to write some code that reads the DXF, organizes the entities into wires, and then outputs them

dcowden, 2018-12-06:
[image] **Comment by [huskier](https://github.com/huskier)**
_Thursday Dec 01, 2016 at 09:11 GMT_

----

Good to know the work plan. If there is something I can help, please fell free to tell me.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Thursday Dec 01, 2016 at 13:17 GMT_

----

I'll definitely be able to use some help. My hope is to describe the way of working through either words or code, and get it flushed out to the point where others (you ;) ) could add functions, tests, etc.  

I have a some time off my day job coming up over the holidays, which is when I plan to do some of this work.

dcowden, 2018-12-06:
[image] **Comment by [dcowden](https://github.com/dcowden)**
_Wednesday Dec 28, 2016 at 21:35 GMT_

----

CQ 2.0 work started

fi0, 2019-11-03:
DXF import would be cool. What's the status and how I can help?

adam-urbanczyk, 2020-05-27:
@fi0 take a look at the OCP-dxf branch

closed 2021-11-11 by marcus7070