CadQuery/cadquery #56: Getting Cadquery to work on Windows 10

JacobEFO, 2018-12-08:
Hey you guys.

I'm trying to figure out a way to use CadQuery on Windows 10. But as far as I can see, CadQuery only supports Python2.6, a version that is no longer possible to get using Anaconda.

Therefore I thought to use Bash on Ubuntu on Windows, but I cannot for the life of me get cadquery to find my Freecad.

Do you guys have any solutions to this issue? Or can I expect to use a newer version of Python in the near future?

Best regards.

adam-urbanczyk, 2018-12-08:
I assume you want to use the FreeCAD based CadQuery. CadQuery as such supports python3. The hard part is to get FreeCAD with python3. If you are familiar/willing to use anaconda you can setup an environment for cadquery as follows:
```
conda create --quiet --name cqtest -c cadquery -c conda-forge freecad=0.17 python=3.6 pyparsing
```
then `conda activate cqtest` and `pip install cadquery`

Let us know if that helps. BTW: other option is to use the PythonOCC based cadquery

JacobEFO, 2018-12-11:
Hi Adam

I will look into this solution. As of right now I've downloaded the CadQuery model directly in freecad and execute Kicad's scripts from a batch file. While it works, it's not a pretty way of doing it :)

Thank you for returning. Currently I've ended up using Anaconda, so it won't be too much of a hassle.

adam-urbanczyk, 2019-02-04:
Closing this issue for now, @JacobEFO  feel free to reopen if CQ is still not working on Windows 10.

closed 2019-02-04 by adam-urbanczyk