CadQuery/cadquery #58: Create a compelling, online-only CQ 2.0 experience

dcowden, 2018-12-10:
When we moved cadquery from to its new home, we lost a lot of stars. But the good news is that we have more to offer-- we just have to get people excited.

I believe the most important thing we can do is to provide an online (no-install) solution, which lets people try cq via a hosted GUI.   This GUI should allow people to try it with no work, play with samples, and (ideally) pull in all of the libraries in the eco-system ( cqparts, cadquery-contrib). 

The result should be fire-up-your browser, and run any of the samples in cadquery-contrib, or cqparts.

I believe the closest we had to his is tryQA, based on jupyter.  It was cool, but it was too slow.

I believe the 'right' direction long-term is to build a javascript-based frontend and a server backend, and then deploy that on a hosted platform.  With these two components, we can then make native apps for all the important platforms using electron/nativify, while using the same GUI code to support hosted installations AND native.  We could even have a mobile app, and the ability for others to easily host.

I'm not long on time, but i'm willing to contribute capital. If we find a solution that works, but just needs a small amount of $ per month for servers, i will do that.

dcowden, 2018-12-10:
@jmwright @adam-urbanczyk your thoughts?

adam-urbanczyk, 2018-12-10:
Definitely a good idea. How about jupyter notebooks and https://mybinder.org/? Not a standalone UI but would be relatively easy to implement.

dcowden, 2018-12-10:
Im ok with that, but i recall that earlier attempts at that were horribly slow. Is there a way to make it fast?

jmwright, 2018-12-12:
Interest in CQ 1.0 didn't really take off until we created the FreeCAD workbench, so I'm all for making sure that we have a good cross-platform way for people to try CQ 2.0.

fragmuffin, 2018-12-12:
I agree that a low barrier to entry will increase the likelihood of people finding cadquery to continue using it (it's a statistical fact; it's impossible to disagree ;) )

A web site is one way, but creating geometry is always going to be a CPU hungry task, so making it fast on a web service will be expensive.
Also: a slow user experience may reflect badly on the library, even though the speed wouldn't be cadquery's fault.

I think the effort is probably better spent on a GUI, like electron (which you mentioned).
Also a spiffy website, can't hurt to be pretty.

@zignig whipped up a great start of a site for cqparts (https://cqparts.github.io)
The HTML and examples are auto-generated, we're currently in the process of cleaning it up and adding more relevant content.
It shows what can be done, and clearly links to the source code that was used to create it... It's frickin' amazing! (and cheap, because static)

jmwright, 2018-12-12:
Wow, I hadn't seen any of the latest iterations of that site. That's awesome! Thanks @zignig !

dcowden, 2018-12-12:
Uh---- holy cow @zignig that's the most amazing thing! its come a long way since i saw it last! its beautiful!

How can we help move this along! its incredible!

zignig, 2018-12-13:
On an away mission until monday, little time. Will laydown a things to be done next week.

@jmwright @dcowden  glad you like it :)

Irev-Dev, 2020-12-28:
@dcowden and I had a bit of a chat recently (plus chatter on the elmer discord) about the possibility of adding CadQuery support into CadHub. I'm definitely open to it. We can discuss more here, though I've added an issue on the Cadhub repo too if that makes more sense for a Cadhub specific integration.
https://github.com/Irev-Dev/cadhub/issues/182

jmwright, 2021-04-23:
@Irev-Dev has made a [preview](https://deploy-preview-277--cadhubxyz.netlify.app/dev-ide#encoded_script=IyAkZm49MzA7CiMgc2NhbGU9MjsKIyByb3RhdGUoWzkwLDAsMF0pY3lsaW5kZXIocj01KnNjYWxlLCBoPTIwKnNjYWxlKTsKCmltcG9ydCBjYWRxdWVyeSBhcyBjcQpmcm9tIGNhZHF1ZXJ5IGltcG9ydCBleHBvcnRlcnMKcmFkaXVzID0gNS4wCnJlc3VsdCA9IChjcS5Xb3JrcGxhbmUoKS5jaXJjbGUocmFkaXVzKS5leHRydWRlKDUwKQogICAgLmZhY2VzKCI+WiIpLndvcmtwbGFuZShpbnZlcnQ9VHJ1ZSkuY2lyY2xlKDEuMDUpLmN1dEJsaW5kKDguMCkKICAgIC5mYWNlcygiPFoiKS53b3JrcGxhbmUoaW52ZXJ0PVRydWUpLmNpcmNsZSgwLjgpLmN1dEJsaW5kKDEyLjApCiAgICAuZWRnZXMoIiVDSVJDTEUiKS5jaGFtZmVyKDAuMTUpKQpzaG93X29iamVjdChyZXN1bHQp) available of CadHub's CadQuery rendering capability. As stated in the banner it's still a work in progress, so keep that in mind. I've spent a little time changing the script and re-rendering the results. If you look at the `encoded_script` parameter in the URL, you'll see that it's possible to embed a script. When this functionality is ready for production use, one of the ways we should be able to use CadHub like a JSFiddle for CadQuery scripts.

jmwright, 2021-04-27:
Here's [a link](https://cadhub.xyz/dev-ide/cadQuery#) to the newer iteration of CadHub's CadQuery support. It's still a development preview, but is already usable for a wide range of models.