CadQuery/cadquery #59: travis-ci doesn't automatically push builds to conda

justbuchanan, 2018-12-12:
Not sure if this is intentional, but I noticed that the linux builds on anaconda.org are ~3 months old, while the builds for Windows reflect the latest successful build on the master branch. Appveyor automatically pushes a build for each passing commit, while travis-ci only seems to push a build for "tagged" commits.

The latest [travis-ci build log](https://travis-ci.org/CadQuery/cadquery/jobs/466646609) shows this message: "Skipping a deployment with the script provider because this is not a tagged commit". Maybe there's a setting for pushing builds on all commits, rather than just ones with tags?

Conda download page: https://anaconda.org/CadQuery/cadquery-occ/files.

adam-urbanczyk, 2019-06-07:
I want to move building to azure and reuse saome of the tamplates in the https://github.com/CadQuery/conda-packages

Background info on reusing templates: https://docs.microsoft.com/en-us/azure/devops/pipelines/process/templates?view=azure-devops

jmwright, 2019-06-07:
@adam-urbanczyk What's the advantage to being on Azure?

adam-urbanczyk, 2019-06-08:
They support all 3 platforms and offer 10 build agents that can run in parallel for up to 6h AFAIR. Also the template support is quite nice as mentioned above.

jmwright, 2019-06-08:
That sounds good.

dcowden, 2019-06-08:
That's really interesting that they support all the platforms. I'd never have guessed. Is it free for open source 'forever' or are they doing some sort of intro/rollout

adam-urbanczyk, 2019-06-10:
@dcowden AFAIK they are free 'forever'. Some serious open source projects are using them: numpy, scipy, spyder. I think they are legit.

dcowden, 2019-06-10:
@adam-urbanczyk awesome! I've been really impressed lately with Microsoft. They were really on a bad path, but VScode is actually really good, and sqlserver is now supported on linux, which amazed me.

I think they're really embracing OS, and it sounds like azure is too!

adam-urbanczyk, 2019-06-10:
Alright azure is working for win and mac but I am having rather esoteric issue regarding fonts on ubuntu.

Some reading material: http://eosrei.net/articles/2016/02/changing-default-fallback-subsitution-fonts-linux

adam-urbanczyk, 2019-06-11:
It works! I had to learn more about fonts and linux than I ever wanted but all is green now.

https://dev.azure.com/cadquery/conda-packages/_build?definitionId=2&_a=summary
https://anaconda.org/CadQuery/cadquery/files

The setup is as follows:
* every commit to master triggers a build that is versioned as `master`
* every tag triggers a build that gets the id of the particular tag

This way one can try a "nightly" build as well a release one. For now I added a `2.0RC0` tag to test this pipeline.

closed 2019-06-11 by adam-urbanczyk