CadQuery/cadquery #60: question - organic modeling

KammutierSpule, 2018-12-14:
I am new to CAD world, zero experience, but as a programmer, cadquery makes sense to me.

I understand the basic features of cadquery and CAD in general as the CSG operations, edge operations, chamfering.. etc
I was looking for a code way to create more organic models.
The result is not meant to be created in life, just displayed.
I learned also a term used in CAD that is "generative design".

Does cadquery allows me to achieve this kind of design?

Can I have things like:
blending between forms?
sweeps with multiple lines controls?

KammutierSpule, 2018-12-14:
Thanks for sharing it.
I just check some operator called "loft" that can generate some interesting results for me.

but here is some screenshots of things I am looking for:
https://help.autodesk.com/cloudhelp/2016/ENU/AutoCAD-Core/images/GUID-3B8A5849-217C-4CCA-8BB0-839F4E28FAFC.png
https://www.dummies.com/wp-content/uploads/435805.image0.jpg
http://docs.autodesk.com/ACD/2010/ENU/AutoCAD%202010%20User%20Documentation/images/PTDCPM/Gator-All/English/ill_22_lofted_solid_text.png

jmwright, 2018-12-14:
I'd like to hear what the other devs think too, but I don't think CadQuery is currently capable of doing what's shown in the first two images. Doing contours will likely require surface manipulation functionality that we don't have at this time. However, now that CadQuery is moving to PythonOCC it opens up some new possibilities for supporting advanced surface operations.

That loft in the last image should be doable though. [Here's an example](https://github.com/dcowden/cadquery/blob/3af828efc155649ad2b8ec7f49b1072699eaaebe/examples/FreeCAD/Ex018_Making_Lofts.py).

dcowden, 2018-12-14:
@KammutierSpule CadQuery (CQ) is a wrapper on top of pythonOCC.  As such, it makes things easier, vs possible.

CQ doesn't have support to make the first two images 'easy' -- as in, a few lines of code.  However, since CQ is based on pythonOCC, you can definitely do those-- you'll just be dropping back down and using pythonOCC directly, vs the CQ wrappers.  

That's ok, because you can mix the two. You can write some pythonOCC code, creat an object, but then use CQ to interact with and enhance that object in subsequent steps. Or, conversely, you can build an object with CQ, and then drop down and use the pythonOCC details to further manipulate it.

KammutierSpule, 2018-12-14:
Thanks for this clarification, it helped me understand "where I am"..

Do you have already some examples on mixing pythonOCC and CQ ?
Could you point me also where I can learn pythonOCC ? I wasn't found a good resource so far...

adam-urbanczyk, 2018-12-20:
Related discussion: https://github.com/tpaviot/pythonocc-core/issues/409

adam-urbanczyk, 2019-01-06:
We also have not the capability to extrude along an arbitrary path and varying sections

Cf. https://github.com/CadQuery/cadquery/pull/65

adam-urbanczyk, 2019-02-11:
@KammutierSpule  I opened #93 , once solved might be also of interest.

KammutierSpule, 2019-02-12:
Hi Adam, thanks for the updates.
I did some experiences but I didn't proceed with anything useful for now.
I used some loft and sweeps but some times it was hard to get it work without crashes. Probably due invalid shape points.
Mainly I ended to experiment more with CSG, unions, intersections, subtract solids... It was ok but once I start more and more solid operations it ended to be extreme slow to process.

Since I am not looking for "mathematical CAD precision", but I am just interested in the triangle/polygon generation output, this may be not the right tool / approach for my proposes.

adam-urbanczyk, 2019-02-13:
Could you share what kind of shapes are you trying to achieve? Maybe I or someone from the community could offer some advice on the best modeling strategy. Also if something crashes and it shouldn't feel free to open an issue.

KammutierSpule, 2019-02-13:
Here are some samples, all generated using CadQuery and CQparts
https://i.imgur.com/332s0QA.png
Exported to gltf and imported with Blender.
... basically some foliage and trunks :)
Different techniques used ( spline, lofts, unions, cut, rotate, translate,...), they use lots of operations.. and takes a while to generate..
Reducing the precision on gltf export helped a bit with the speed.

jmwright, 2019-12-20:
@KammutierSpule This may not be exactly what you're looking for, but [this PR](https://github.com/CadQuery/cadquery/issues/229) will enable more organic shapes to be created when it's merged.