CadQuery/cadquery #62: show_object does not support option parameter

tolot27, 2018-12-17:
In QC1.2 show_object allows to pass an option parameter, i. e. to set the color and transparency of the shape (`show_object(result, options={"rgba":(128, 255, 128, 0.0)})`). Unfortunately, this parameter is not available in cadquery-occ and the following error occurs:

```
TypeError: () got an unexpected keyword argument 'options'
```

I know, there are likely still a lot of incompatibilities yet between QC1.2 and QC-occ. Will they be solved in the future such as a direct upgrade without much code adjustments would be possible?

adam-urbanczyk, 2018-12-20:
Yes, that is definitely our ambition

adam-urbanczyk, 2019-02-04:
Should be fixed by #86. @jmwright  could you confirm?

jmwright, 2019-02-04:
Yes, when #86 is merged it should fix this. It's still WIP for now though. I'm struggling to get an environment set up so I can run the test suite locally, but after I figure that out the progress on this should be pretty quick.

jmwright, 2019-02-07:
@tolot27 If you're still watching this issue, this should be fixed in the adam-urbanczyk-OCC-version-update branch.

If you're interested in trying it, let us know if you want any help getting set up.

jmwright, 2019-02-10:
Closing this. The fix isn't in master yet, but is accessible now on a branch and will be in master at some point.

closed 2019-02-10 by jmwright

tolot27, 2019-06-25:
@adam-urbanczyk Can you merge the fix into master? The branch is no longer available.

tolot27, 2019-06-25:
See [CQ-editor doesn't recognize show_object() options variable](https://groups.google.com/forum/#!msg/cadquery/eQbwXNyiQ7U/hLXkWe6sCgAJ)