CadQuery/cadquery #63: Comment in Patreon Update About PyOCCT Support

jmwright, 2018-12-19:
tl;dr Kurt makes a quick case for using PyOCCT in CadQuery instead of PythonOCC in [this post](https://www.patreon.com/posts/my-activities-in-23424216) ("cadquery" section).

There's also a "PyOCCT" section a ways below that that gives a little more background.

I'm not pushing for changing course at this point, but I've had an email conversation with Kurt about his work and about CadQuery. He really seems to be tied in with what's going on in the open source science and engineering software world. I appreciate him adding CadQuery to his packaging wishlist and for giving thoughtful feedback. I think it's good to have all information and viewpoints in front of us, even if we don't do anything different. Depending on the situation, maybe it doesn't have to be an either-or thing anyway? I'm not sure how different PythonOCC and PyOCCT are in implementation.

jmwright, 2018-12-19:
Here's [a related conversation](https://github.com/dcowden/cadquery/issues/268#issuecomment-390688427) from the CQ 1.0 issue tracker.

adam-urbanczyk, 2018-12-20:
Interesting @jmwright . I'd love to use OCC 7.3+ under the hood . Actually there is not that much difference between PythonOCC  and PyOCCT API - they are both roughly 1:1 translation of the C++ API of OCC.

The issue for now is that PyOCCT bindings are generated semi-automatically (i.e. partially by hand). It would be highly preferable that we have  fully automatically generated binding so that once future versions of OCC are released there is minimal work required to use them.

NB: I started to work on such a binding generator (heavily inspired by the PyOCCT project) here: https://github.com/CadQuery/cpp-py-bindgen but it is not finished yet.

dcowden, 2018-12-20:
Yep I was also turned off by the hand generated bindings when I looked.

adam-urbanczyk, 2019-06-13:
Closing this. Related issue: #114. In the near future I'd like to finalize https://github.com/CadQuery/cpp-py-bindgen and asses how realistic it is to use our own bindings.

closed 2019-06-13 by adam-urbanczyk