CadQuery/cadquery #69: 2018 Year-End Update

jmwright, 2018-12-31:
## 2018 Year-End Update

If you got a notification because you were mentioned in this issue and are not sure why, scroll down to the "Thank You" section. You are probably mentioned there. I apologize if anyone feels spammed, but the purpose is to say thank you for being involved in this project.

### Intro

CadQuery has seen another exciting year as the community has continued to grow. 2018 has seen the creation of the CadQuery organization on GitHub, to allow leadership responsibilities for the project to be distributed more easily. The GitHub organization has 7 members already, which is a big step up from the days when there were just two of us. Having an organization with a larger leadership component is an important aspect of sustainability, and it is great to see so many people applying their talents to the project.

The creation of the organization allowed repositories that were previously maintained separately to be brought under one umbrella. The biggest repository transition was [cadquery](https://github.com/dcowden/cadquery), which was managed under the account of @dcowden previously. The original repository still exists, but has a note that the transition is underway. The community took the opportunity to port the repository to PythonOCC, resulting in the versioning scheme being bumped up to 2.0. [CadQuery 2.0](https://github.com/CadQuery/cadquery) (aka CQ 2.0) isn't quite ready to take the place of 1.0 (aka CQ 1.0), but it's moving that way quickly thanks to the work of @adam-urbanczyk and @justbuchanan . All their work on the 2.0 transition is greatly appreciated.

The user base has continued to grow as well, with many users actively involved in asking questions and making feature requests. The original CadQuery repo now has almost 400 stars and I know @dcowden is sad to lose those during this 2.0 transition. If you have starred the original CQ 1.0 repository but haven't starred the new one yet, please take some time to do that.

The core CadQuery library is not the only thing that the community has going on. Between last year and this year, multiple new projects were introduced that have improved the CadQuery ecosystem. Below is a summary of some of the community's existing and new activities.

### Projects

- [cadquery](https://github.com/CadQuery/cadquery) - As mentioned above, this core library that we all know and love is headed towards PythonOCC capability, which will be CQ 2.0. This will give us a great deal of flexibility and power in the features that can be added in the future. @adam-urbanczyk has been leading the transition to CQ 2.0, and we are grateful for all his time and effort.
- [cqparts](https://cqparts.github.io/cqparts/doc/) -  cqparts is the creation of @fragmuffin and is a wrapper around cadquery that allows for more complex parametric models to be created, including parts and assemblies. The cqparts project has its own organization now as well, with an ever-growing user base. A CQ 2.0 branch is being worked on [here](https://github.com/CadQuery/cqparts). If you are a CadQuery user, it is highly recommended that you check out cqparts.
- [CQ-editor](https://github.com/CadQuery/CQ-editor) - This is a stand-alone CadQuery IDE created by @adam-urbanczyk . It has CQ  2.0 compatibility, and has some advanced features like a graphical debugger and an object stack inspector. The debugger allows users to step through a script to watch how the model changes with each line. At this time CQ-editor is Linux-only, but Windows and MacOS support are being worked on.
- [cadquery-freecad-module](https://github.com/jmwright/cadquery-freecad-module) - This is a FreeCAD workbench that adds a code editor and some debugging features. Early 2019 will see a major update to this workbench, with a new custom built-in editor and tighter integration for users who want to use external code editors like vim, emacs, VS Code, etc. The future of this workbench is a little uncertain in light of the CQ 2.0 transition. Feedback from the community is welcome as we decide if this workbench still makes sense in the new CQ 2.0 era.
- Web interfaces - Community member @zignig has been doing some work on creating web interfaces for cqparts models, which has culminated in the [cqparts website](https://cqparts.github.io/). I would highly recommend playing around on the site, especially the _Parts Library_ under [Examples](https://cqparts.github.io/#examples). @zignig also created a cqparts [web API](https://github.com/cqparts/cqparts_webapi) and a [stand-alone server](https://github.com/zignig/cqparts-server) for displaying your cqparts models in a web browser locally.

If I have missed any projects, please feel free to reply with their names.

### Thank You

As always, a big _Thank You_ goes out to all our contributors and users. It's exciting to watch the community grow, and to see the creative ways people use CadQuery. I'm going to make an attempt at listing all the 2018 contributors here, but with many contributions spread across several repos in multiple projects I run the risk of missing someone. I apologize if that happens.

#### CadQuery
- @adam-urbanczyk
- @fragmuffin
- @justbuchanan
- @RustyVermeer
- @bweissinger
- @asukiaaa
- @osterwood
- @Grawp
- @johnbeard
- @jduquennoy
- @luzpaz
- @bollwyvl
- @paralax
- @gntech
- @jpmlt
- @dcowden
- @jmwright

#### cqparts
- @fragmuffin
- @zignig
- @gntech
- @justbuchanan
- @luzpaz

#### CQ-editor
- @adam-urbanczyk
- @justbuchanan
- @jmwright

#### cadquery-freecad-module
- @luzpaz
- @fragmuffin
- @jmwright

fragmuffin, 2019-01-01:
Thanks for doing this @jmwright !
It's great to see what we can achieve in a year outside our day-to-day schedules.
I feel motivated for another year of collaboration with all of you wonderful people!

fragmuffin, 2019-01-03:
**Map of Contributors**

I was curious about where all cadquery contributors are from, so for those of you that have added a location to your github profile, I made this map:

[image]
https://www.zeemaps.com/map?group=3278974

I thought it was cool how cadquery is attracting people from all around the world :blush: