CadQuery/cadquery #70: Extrude / cut with draft angle

osterwood, 2018-12-31:
It would be wonderful if extrude and cut operations had a key-word argument for draft angle.  This would allow positive or negative rake on the profile of the operation.

I know this is a tough enhancement to implement -- but it would be very useful to design for some kinds of manufacturing processes.  PythonOCC-core has a few references to draft angle in the codebase, but use is unclear to me.

dcowden, 2018-12-31:
this might be an interesting opportunity to compare pythonOCC and pyOCCT .  There has been some talk of using pyOCCT instead of pythonOCC.   If pyOCCT proved to have clearer and/or better documented interfaces to the draft features, it might bode well for it vs pythonOCC.

osterwood, 2019-01-01:
I’ll have to play around with that.  Thanks.  

Any idea how it would / could work with a blind cavity (e.g. not a thru cut into an existing solid)?

adam-urbanczyk, 2019-01-02:
@osterwood  take a look at #71 .

For now only low level function in `occ_impl`. Do you have any thoughts on API for cq? E.g.

* OCC kernel does not support tapers (at least the function I am using - feel free to browse thorough sources of FreeCAD or SALOME to find something better) with multiple base wires
* Only normal (or -normal) direction is supported

Regarding your question - I am not sure if I get what you mean. Could you paste a sketch?

adam-urbanczyk, 2019-01-06:
@osterwood any luck with #71 ? Is it going in the right direction for your use case?

adam-urbanczyk, 2019-02-04:
@osterwood did you have some time to try out my proposal?

adam-urbanczyk, 2019-02-07:
I added `extrude` and `cutBlind` with taper in #71. Closing this issue

closed 2019-02-07 by adam-urbanczyk

jmwright, 2019-02-07:
Thanks @adam-urbanczyk !