CadQuery/cadquery #75: Add Codacy to CI?

fragmuffin, 2019-01-03:
I recently added [codacy](https://www.codacy.com/) to the `cqparts` CI chain for PRs, and I think it's going quite well.

- What do you think about adding it to `cadquery`s CI?
- Can you suggest a better alternative?

**First Impressions / First Installing**
When first directed codacy toward the `cqparts` repo', it scared me with a terribly low quality score.
But I quickly realised that most of that is knit-picking my less strict formatting of `.md` files, `.js`, `.css`, and other files that aren't really the main "product".

However, that's definitely a worthwhile project in the future: going through the lists of warnings, and either fixing the code, or manually checking them off as OK.

**Advantage for PRs**
It will inform you of unused libraries, bad formatting, and potential security problems.
I tried adding something like:

```python
import subprocess, os
subprocess.Popen(os.environ['foo'])
```

And it failed the test, warning me that this addition could be dangerous.
Then once I added a new commit with those changes removed, the test passed again... it's nifty.

Output is here (while it's still available): https://app.codacy.com/app/fragmuffin/cqparts/commit?bid=10377018&cid=302653800 ; I'm assuming it'll be deleted after some time 'lapses

jmwright, 2019-01-13:
@fragmuffin Does GitHub use Codacy as a check before merging PRs?

fragmuffin, 2019-01-16:
@jmwright 

It does a check per commit to a PR, only running a check on what will be merged; it's just like TravisCI or AppVeyor, but the setup is much simpler.

So for example, the PR cqparts/cqparts#141 for commit [`93375c3`](https://github.com/cqparts/cqparts/pull/141/commits/93375c301676ac69952551c479564244c7fd9e25) has CI tasks listed as:
[image]

Link to the Codacy [Details](https://app.codacy.com/app/fragmuffin/cqparts/pullRequest?prid=2805316)

When it's a green tick, the information isn't that useful, it's just a repeat of what's on github, but for failures, I think it gives a good list of the "failures", and suggestions on how to fix.

It could just be added for a trial, then removed if we don't like it... that's what I'm doing with cqparts, but so far I it's been useful, so I think it's staying.