CadQuery/cadquery #76: uuid1 broken on maocs for python 2.7

bernhard-42, 2019-01-04:
cadquery/occ_impl/jupyter_tools.py uses `uud1` in line 50

In macos this is broken for python 2.7 (tested with anaconda python 2.7.15), see https://bugs.python.org/issue32502
Using `uuid4` would work on all versions of python on the mac (and all other systems, I guess)

fragmuffin, 2019-01-04:
Nice find @bernhard-42

**Some Links**
Line in question
https://github.com/CadQuery/cadquery/blob/568ee63e1a6850f5c63494f0c272c67d2ccd5792/cadquery/occ_impl/jupyter_tools.py#L50

[uuid docs](https://docs.python.org/3/library/uuid.) state:

> ### uuid.uuid1(node=None, clock_seq=None)
> Generate a UUID from a host ID, sequence number, and the current time. If node is not given, getnode() is used to obtain the hardware address. If clock_seq is given, it is used as the sequence number; otherwise a random 14-bit sequence number is chosen.
> ### uuid.uuid4()
> Generate a random UUID.

I didn't even know the `uuid` library existed, so I'm learning.

**Support for python 2.x?**
I think we've previously agreed that the `cadquery` will only be supporting python versions `>=3.5` ([comment is here](https://github.com/dcowden/cadquery/issues/275#issuecomment-396055788))

However, with that said, if `uuid1()` is fallible on some OS's, `uuid4()` seems like a better option anyway.

bernhard-42, 2019-01-04:
I fully agree, python 3 is the way forward.

When I install cadquery 2 (with the conda command in an empty conda python 3 env) it works.
However as soon as I do `conda install jupyter` then in ipython I see:

```
Python 3.6.8 |Anaconda, Inc.| (default, Dec 29 2018, 19:04:46)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.2.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import cadquery as cq
Fatal Python error: PyThreadState_Get: no current thread

Abort trap: 6
```

Hence I ended up in python 2.7 (replacing uud1 with uuid4 was easier)

I will open an issue for this error

bernhard-42, 2019-01-04:
For the python 3.6 issue, see #78

adam-urbanczyk, 2019-01-06:
Solved by #77

closed 2019-01-06 by adam-urbanczyk