CadQuery/cadquery #78: After installing ipython/jupyter cadquery crashes on macos under Python 3.6

bernhard-42, 2019-01-04:
Somehow the installation of ipython kills cadquery:

I use miniconda as base with conda 4.5.2

1) Install conda environment and cadquery-occ

```
$ conda create -n py3occ python=3.6
$ source activate py3occ
$ conda install -c pythonocc -c oce -c conda-forge -c dlr-sc -c CadQuery cadquery-occ

```

2) Test

```
$ python

Python 3.6.6 | packaged by conda-forge | (default, Jul 26 2018, 09:55:02)
[GCC 4.2.1 Compatible Apple LLVM 6.1.0 (clang-602.0.53)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery
>>>
```

3) Install ipython

```
$ conda install ipython

Solving environment: done

## Package Plan ##

  environment location: /opt/miniconda/envs/py3occ

  added / updated specs:
    - ipython

The following NEW packages will be INSTALLED:

    appnope:          0.1.0-py36hf537a9a_0
    backcall:         0.1.0-py36_0
    decorator:        4.3.0-py36_0
    ipython:          7.2.0-py36h39e3cac_0
    ipython_genutils: 0.2.0-py36h241746c_0
    jedi:             0.13.2-py36_0
    parso:            0.3.1-py36_0
    pexpect:          4.6.0-py36_0
    pickleshare:      0.7.5-py36_0
    prompt_toolkit:   2.0.7-py36_0
    ptyprocess:       0.6.0-py36_0
    pygments:         2.3.1-py36_0
    six:              1.12.0-py36_0
    traitlets:        4.3.2-py36h65bd3ce_0
    wcwidth:          0.1.7-py36h8c6ec74_0

The following packages will be UPDATED:

    openssl:          1.0.2p-h470a237_2     conda-forge --> 1.1.1a-h1de35cc_0
    python:           3.6.6-h5001a0f_0      conda-forge --> 3.6.8-haf84260_0

The following packages will be DOWNGRADED:

    ca-certificates:  2018.11.29-ha4d7672_0 conda-forge --> 2018.03.07-0
    certifi:          2018.11.29-py36_1000  conda-forge --> 2018.11.29-py36_0
```

4) Test

```
$ ipython
Python 3.6.8 |Anaconda, Inc.| (default, Dec 29 2018, 19:04:46)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.2.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import cadquery as cq
Fatal Python error: PyThreadState_Get: no current thread

Abort trap: 6
```

jmwright, 2019-01-11:
Not sure if this is related, but I see this error in the CQ-editor console when running it with the latest updates on MacOS.
```
/Users/User1/miniconda3/envs/cqgui/lib/python3.6/site-packages/jupyter_client/session.py:803: DeprecationWarning: Accessing zmq Socket attribute recv_multipart on BackgroundSocket
  msg_list = socket.recv_multipart(mode, copy=copy)
``

adam-urbanczyk, 2019-01-12:
@bernhard-42 could you also paste a backtrace from the crash?

bernhard-42, 2019-01-12:
Unfortunately, there is nothing more than the three lines above from "Fatal" to "trap: 6"

However, I found out that when you reinstall cadquery with the
`conda install -c pythonocc -c oce -c conda-forge -c dlr-sc -c CadQuery cadquery-occ`
standard command (reproducible) cadquery works fine again.

It looks like the issue comes from python (whether it is from anaconda or from python-forge).

Steps to reproduce:

1) Install ipython again while cadquery is working under python 3.6.6 from conda-forge:

```
$ conda install ipython
Solving environment: done

## Package Plan ##

  environment location: /opt/miniconda3/envs/pycq

  added / updated specs:
    - ipython

The following packages will be UPDATED:

    openssl:         1.0.2p-h470a237_2     conda-forge --> 1.1.1a-h1de35cc_0
    python:          3.6.6-h5001a0f_0      conda-forge --> 3.6.8-haf84260_0

The following packages will be DOWNGRADED:

    ca-certificates: 2018.11.29-ha4d7672_0 conda-forge --> 2018.03.07-0
    certifi:         2018.11.29-py36_1000  conda-forge --> 2018.11.29-py36_0

Proceed ([y]/n)?

Preparing transaction: done
Verifying transaction: done
Executing transaction: done

```

We now have anaconda python and conda-forge python is removed.

Result: cadquery crashes

```
$ ipython
Python 3.6.8 |Anaconda, Inc.| (default, Dec 29 2018, 19:04:46)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.2.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import cadquery
Fatal Python error: PyThreadState_Get: no current thread

Abort trap: 6
```

2) Let's reinstall cadquery:

```
$ conda install -c pythonocc -c oce -c conda-forge -c dlr-sc -c CadQuery cadquery-occ
Solving environment: done

## Package Plan ##

  environment location: /opt/miniconda3/envs/pycq

  added / updated specs:
    - cadquery-occ

The following packages will be UPDATED:

    ca-certificates: 2018.03.07-0      --> 2018.11.29-ha4d7672_0 conda-forge
    certifi:         2018.11.29-py36_0 --> 2018.11.29-py36_1000  conda-forge

The following packages will be DOWNGRADED:

    openssl:         1.1.1a-h1de35cc_0 --> 1.0.2p-h470a237_2     conda-forge
    python:          3.6.8-haf84260_0  --> 3.6.6-h5001a0f_0      conda-forge

Proceed ([y]/n)?

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
```

Now we get conda-forge python again, and caqdquery works again.

```
$ ipython
Python 3.6.6 | packaged by conda-forge | (default, Jul 26 2018, 09:55:02)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.2.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import cadquery

In [2]:
```

This is in line with https://github.com/conda-forge/gdcm-feedstock/issues/8#issuecomment-414716548

What I don't know is whether the build process of conda-forge is different or version 3.6.6 vs. 3.6.8 (and I have no idea how to check it)

adam-urbanczyk, 2019-01-14:
@bernhard-42 could you generate a core dump and inspect it. *Roughly* like this:
```
ulimit -c unlimited -S

# reproduce the error

#  find the latest core file: /cores/core.*

lldb --core $PATH_TO_CORE_FILE --batch --one-line "bt" 
```

bernhard-42, 2019-01-15:
Here you are:

```
$ lldb --core /cores/core.16046 --batch --one-line "bt"

(lldb) target create --core "/cores/core.16046"
Core file '/cores/core.16046' (x86_64) was loaded.
(lldb) bt
* thread #1, stop reason = signal SIGSTOP
  * frame #0: 0x00007fff6e64623e libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff6e6fcc1c libsystem_pthread.dylib`pthread_kill + 285
    frame #2: 0x00007fff6e5af1c9 libsystem_c.dylib`abort + 127
    frame #3: 0x00000001070aed23 libpython3.6m.dylib`Py_FatalError + 371
    frame #4: 0x0000000106fc984b libpython3.6m.dylib`PyModule_Create2 + 699
    frame #5: 0x0000000106c56e49 _gp.so`PyInit__gp + 601
    frame #6: 0x0000000104872e99 python`_PyImport_LoadDynamicModuleWithSpec + 473
    frame #7: 0x0000000104872523 python`_imp_create_dynamic + 211
    frame #8: 0x0000000104771a82 python`PyCFunction_Call + 66
    frame #9: 0x0000000104845474 python`_PyEval_EvalFrameDefault + 47604
    frame #10: 0x00000001048388c9 python`_PyEval_EvalCodeWithName + 425
    frame #11: 0x0000000104847a8a python`fast_function + 362
    frame #12: 0x000000010484767c python`call_function + 236
    frame #13: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #14: 0x00000001048479dc python`fast_function + 188
    frame #15: 0x000000010484767c python`call_function + 236
    frame #16: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #17: 0x00000001048479dc python`fast_function + 188
    frame #18: 0x000000010484767c python`call_function + 236
    frame #19: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #20: 0x00000001048479dc python`fast_function + 188
    frame #21: 0x000000010484767c python`call_function + 236
    frame #22: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #23: 0x00000001048479dc python`fast_function + 188
    frame #24: 0x000000010484767c python`call_function + 236
    frame #25: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #26: 0x00000001048479dc python`fast_function + 188
    frame #27: 0x000000010484767c python`call_function + 236
    frame #28: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #29: 0x00000001048388c9 python`_PyEval_EvalCodeWithName + 425
    frame #30: 0x0000000104847a8a python`fast_function + 362
    frame #31: 0x000000010484767c python`call_function + 236
    frame #32: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #33: 0x00000001048388c9 python`_PyEval_EvalCodeWithName + 425
    frame #34: 0x0000000104847a8a python`fast_function + 362
    frame #35: 0x000000010484767c python`call_function + 236
    frame #36: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #37: 0x00000001048479dc python`fast_function + 188
    frame #38: 0x000000010484767c python`call_function + 236
    frame #39: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #40: 0x00000001048388c9 python`_PyEval_EvalCodeWithName + 425
    frame #41: 0x0000000104833129 python`builtin_exec + 345
    frame #42: 0x0000000104771a82 python`PyCFunction_Call + 66
    frame #43: 0x0000000104845474 python`_PyEval_EvalFrameDefault + 47604
    frame #44: 0x00000001048388c9 python`_PyEval_EvalCodeWithName + 425
    frame #45: 0x0000000104847a8a python`fast_function + 362
    frame #46: 0x000000010484767c python`call_function + 236
    frame #47: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #48: 0x00000001048479dc python`fast_function + 188
    frame #49: 0x000000010484767c python`call_function + 236
    frame #50: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #51: 0x00000001048479dc python`fast_function + 188
    frame #52: 0x000000010484767c python`call_function + 236
    frame #53: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #54: 0x00000001048479dc python`fast_function + 188
    frame #55: 0x000000010484767c python`call_function + 236
    frame #56: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #57: 0x0000000104848212 python`_PyFunction_FastCallDict + 178
    frame #58: 0x00000001046f0f80 python`_PyObject_FastCallDict + 320
    frame #59: 0x00000001046fa1ba python`_PyObject_CallMethodIdObjArgs + 490
    frame #60: 0x000000010486c804 python`PyImport_ImportModuleLevelObject + 1092
[... 72 lines cut ...]
    frame #133: 0x00000001048388c9 python`_PyEval_EvalCodeWithName + 425
    frame #134: 0x0000000104847a8a python`fast_function + 362
    frame #135: 0x000000010484767c python`call_function + 236
    frame #136: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #137: 0x00000001048479dc python`fast_function + 188
    frame #138: 0x000000010484767c python`call_function + 236
    frame #139: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #140: 0x00000001048388c9 python`_PyEval_EvalCodeWithName + 425
    frame #141: 0x00000001048482cc python`_PyFunction_FastCallDict + 364
    frame #142: 0x00000001046f0f80 python`_PyObject_FastCallDict + 320
    frame #143: 0x00000001047185f8 python`method_call + 136
    frame #144: 0x00000001046f85ce python`PyObject_Call + 62
    frame #145: 0x0000000104845376 python`_PyEval_EvalFrameDefault + 47350
    frame #146: 0x00000001048388c9 python`_PyEval_EvalCodeWithName + 425
    frame #147: 0x0000000104847a8a python`fast_function + 362
    frame #148: 0x000000010484767c python`call_function + 236
    frame #149: 0x0000000104845175 python`_PyEval_EvalFrameDefault + 46837
    frame #150: 0x00000001048388c9 python`_PyEval_EvalCodeWithName + 425
    frame #151: 0x000000010489155c python`PyRun_FileExFlags + 252
    frame #152: 0x0000000104890a34 python`PyRun_SimpleFileExFlags + 372
    frame #153: 0x00000001048b77c6 python`Py_Main + 3734
    frame #154: 0x00000001046e8f59 python`main + 313
    frame #155: 0x00007fff6e506ed9 libdyld.dylib`start + 1
    frame #156: 0x00007fff6e506ed9 libdyld.dylib`start + 1
```

bernhard-42, 2019-01-15:
could it have anything to do with https://github.com/tpaviot/pythonocc-core/issues/541 ?

adam-urbanczyk, 2019-01-15:
@bernhard-42 thanks, yes it has to be it. I uploaded the latest build  of pythonocc 
(that should solve this issue) to the cadquery channel. Could you update your env and confirm that it works?

bernhard-42, 2019-01-16:
Well, now it doesn't work any more, even if you install python-cadquery again. Looks like another build of python 3.6.6 is in conda-forge (Oct 2018 instead July 2018).

```
$ conda create -n pycq2 python=3.6 ipython jupyter notebook ipywidgets
...
$ conda install -c pythonocc -c oce -c conda-forge -c dlr-sc -c CadQuery cadquery-occ
...
$ ipython
Python 3.6.6 | packaged by conda-forge | (default, Oct 12 2018, 07:24:56)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.2.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import cadquery
Fatal Python error: PyThreadState_Get: no current thread

Abort trap: 6
```

I now have to do a to get rid of python 3.6.6 and use python 3.6.5

```
$ conda install -c conda-forge python=3.6.5
$ $ ipython
Python 3.6.5 | packaged by conda-forge | (default, Apr  6 2018, 13:44:09)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.2.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import cadquery

In [2]:
```
Something is wrong in 3.6.**6**

adam-urbanczyk, 2019-01-16:
@bernhard-42 well I assume it is this static linking thing. Could you try explicitly with the latest build (NB: an experimental build from "our" channel): `conda install -c cadquery pythonocc=0.18.2=py36hcda10a6_0`?

bernhard-42, 2019-01-17:
Not too much of a success:

```
$conda create -n pycq2 python=3.6 jupyter ipywidgets
$ conda install -c cadquery pythonocc=0.18.2=py36hcda10a6_0
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - pythonocc==0.18.2=py36hcda10a6_0

Current channels:

  - https://conda.anaconda.org/cadquery/osx-64
  - https://conda.anaconda.org/cadquery/noarch
  - https://repo.anaconda.com/pkgs/main/osx-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/free/osx-64
  - https://repo.anaconda.com/pkgs/free/noarch
  - https://repo.anaconda.com/pkgs/r/osx-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/pro/osx-64
  - https://repo.anaconda.com/pkgs/pro/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

adam-urbanczyk, 2019-01-17:
@bernhard-42 I made a mistake. It is `pythonocc-core` not `pythonocc`. So it should be `conda install -c cadquery pythonocc-core=0.18.2=py36hcda10a6_0`

bernhard-42, 2019-01-17:
I looked at `https://conda.anaconda.org/cadquery/osx-64` and already tried it.
Same result. Digging deeper into it, one also needs `-c dlr-sc` so that conda finds `freeimageplus`

So the full install is:

`conda install -c dlr-sc -c cadquery pythonocc-core=0.18.2=py36hcda10a6_0`

A good sign is that conda does not try to install a python interpreter from conda-forge any more.

```
$ conda create -n pycq3 python=3.6
$ conda install -c dlr-sc -c cadquery pythonocc-core=0.18.2=py36hcda10a6_0
Solving environment: done

## Package Plan ##

  environment location: /opt/miniconda3/envs/pycq3

  added / updated specs:
    - pythonocc-core==0.18.2=py36hcda10a6_0

The following NEW packages will be INSTALLED:

    freeimageplus:  3.17.0-1              dlr-sc
    freetype:       2.9.1-hb4e5f40_0
    gl2ps:          1.3.8-0               dlr-sc
    libpng:         1.6.36-ha441bb4_0
    oce:            0.18.3-he0a9f98_0     cadquery
    pythonocc-core: 0.18.2-py36hcda10a6_0 cadquery
    six:            1.12.0-py36_0
    tbb:            2019.1-h04f5b5a_0

Proceed ([y]/n)?
```

Trying to install `cadquery-occ` on top failed sine it requires `pythonocc-core=0.18.1`. so my usual test `import cadquery as cq` is not possible. However in my broken environment `import OCC.Core.TopAbs` fails and in the installation done with the command above it work.

Even installing ipython afterwards keeps OCC working.

I guess we now need a cadquery version working with occ 0.18.2 to finalize the test

bernhard-42, 2019-01-19:
works fine!

btw. the syntax I had to use is
`pip install git+https://github.com/CadQuery/cadquery.git@adam-urbanczyk-OCC-version-update`

bernhard-42, 2019-01-19:
I guess we can close the issue now?

adam-urbanczyk, 2019-01-19:
Great! Let's wait until this is in master branch.

bernhard-42, 2019-01-19:
Without gdb, it fails in `fillet`:

````
In [1]: import cadquery as cq

In [2]: b = cq.Workplane('XY')

In [3]: box1 = b.box(1,2,3).edges(">X").chamfer(0.1)

In [4]: cq.Vector(0, 1.5, 0.8)
Out[4]: Vector: (0.0, 1.5, 0.8)

In [5]: b.transformed(offset=cq.Vector(0, 1.5, 0.8))
Out[5]: <cadquery.cq.Workplane at 0x117f6fe10>

In [6]: b.transformed(offset=cq.Vector(0, 1.5, 0.8)).box(3,2,1)
Out[6]: <cadquery.cq.Workplane at 0x117fb7400>

In [7]: b.transformed(offset=cq.Vector(0, 1.5, 0.8)).box(3,2,1).edges(">Z")
Out[7]: <cadquery.cq.Workplane at 0x117f6fd30>

In [8]: b.transformed(offset=cq.Vector(0, 1.5, 0.8)).box(3,2,1).edges(">Z").fil
   ...: let(0.1)
```

bernhard-42, 2019-01-19:
gdb not sufficiently code-signed on my mac, what so ever this means. No idea how to get that running. I'm a python, javascript and scala guy, gcc and gdb are strangers to me ;-)

adam-urbanczyk, 2019-01-19:
Maybe lldb then? I don't have Mac experience...

bernhard-42, 2019-01-19:
This is the current threads call stack
It seems to be the same every time I Ctrl-C
```
bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
  * frame #0: 0x00007fff75aa526a libsystem_malloc.dylib`free_tiny + 648
    frame #1: 0x000000010583d39a libTKMath.11.dylib`math_DoubleTab::Free() + 26
    frame #2: 0x00000001052845e8 libTKGeomBase.11.dylib`AppCont_LeastSquare::AppCont_LeastSquare(AppCont_Function const&, double, double, AppParCurves_Constraint, AppParCurves_Constraint, int, int) + 10792
    frame #3: 0x00000001052516a4 libTKGeomBase.11.dylib`Approx_FitAndDivide2d::Compute(AppCont_Function const&, double, double, double&, double&) + 196
    frame #4: 0x0000000105251307 libTKGeomBase.11.dylib`Approx_FitAndDivide2d::Perform(AppCont_Function const&) + 487
    frame #5: 0x00000001052510a3 libTKGeomBase.11.dylib`Approx_FitAndDivide2d::Approx_FitAndDivide2d(AppCont_Function const&, int, int, double, double, bool, AppParCurves_Constraint, AppParCurves_Constraint) + 227
    frame #6: 0x000000010521936a libTKGeomBase.11.dylib`ProjLib_ComputeApprox::ProjLib_ComputeApprox(Handle_Adaptor3d_HCurve const&, Handle_Adaptor3d_HSurface const&, double) + 2362
    frame #7: 0x0000000105237f38 libTKGeomBase.11.dylib`ProjLib_ProjectedCurve::Load(Handle_Adaptor3d_HCurve const&) + 4056
    frame #8: 0x00000001052391af libTKGeomBase.11.dylib`ProjLib_ProjectedCurve::ProjLib_ProjectedCurve(Handle_Adaptor3d_HSurface const&, Handle_Adaptor3d_HCurve const&, double) + 111
    frame #9: 0x000000010523cf42 libTKGeomBase.11.dylib`GeomProjLib::Curve2d(Handle_Geom_Curve const&, double, double, Handle_Geom_Surface const&, double, double, double, double, double&) + 482
    frame #10: 0x0000000104cc8716 libTKGeomAlgo.11.dylib`GeomInt_IntSS::BuildPCurves(double, double, double&, Handle_Geom_Surface const&, Handle_Geom_Curve const&, Handle_Geom2d_Curve&) + 198
    frame #11: 0x0000000104cc484a libTKGeomAlgo.11.dylib`GeomInt_IntSS::MakeCurve(int, Handle_Adaptor3d_TopolTool const&, Handle_Adaptor3d_TopolTool const&, double, bool, bool, bool) + 6746
    frame #12: 0x0000000104cc25c3 libTKGeomAlgo.11.dylib`GeomInt_IntSS::InternalPerform(double, bool, bool, bool, bool, double, double, double, double) + 435
    frame #13: 0x0000000103e6ce4f libTKFillet.11.dylib`ChFi3d_ComputeCurves(Handle_Adaptor3d_HSurface&, Handle_Adaptor3d_HSurface&, TColStd_Array1OfReal const&, TColStd_Array1OfReal const&, Handle_Geom_Curve&, Handle_Geom2d_Curve&, Handle_Geom2d_Curve&, double, double, double&, bool) + 1631
    frame #14: 0x0000000103ec1ec4 libTKFillet.11.dylib`ChFi3d_Builder::PerformTwoCornerbyInter(int) + 4388
    frame #15: 0x0000000103f0df44 libTKFillet.11.dylib`ChFi3d_FilBuilder::PerformTwoCorner(int) + 4484
    frame #16: 0x0000000103e5bdb0 libTKFillet.11.dylib`ChFi3d_Builder::PerformFilletOnVertex(int) + 736
    frame #17: 0x0000000103e59ed2 libTKFillet.11.dylib`ChFi3d_Builder::Compute() + 1570
    frame #18: 0x0000000103fe1f1e libTKFillet.11.dylib`BRepFilletAPI_MakeFillet::Build() + 30
    frame #19: 0x0000000104a8e78c libTKTopAlgo.11.dylib`BRepBuilderAPI_MakeShape::Shape() const + 28
    frame #20: 0x000000010852e034 _BRepBuilderAPI.so`_wrap_BRepBuilderAPI_MakeShape_Shape(_object*, _object*) + 212
    frame #21: 0x000000010008a0e2 python`_PyCFunction_FastCallDict + 322
    frame #22: 0x0000000100011171 python`_PyObject_FastCallKeywords + 385
    frame #23: 0x000000010015f718 python`call_function + 392
    frame #24: 0x000000010015d175 python`_PyEval_EvalFrameDefault + 46837
    frame #25: 0x000000010015f9dc python`fast_function + 188
    frame #26: 0x000000010015f67c python`call_function + 236
    frame #27: 0x000000010015d175 python`_PyEval_EvalFrameDefault + 46837
    frame #28: 0x000000010015f9dc python`fast_function + 188
    frame #29: 0x000000010015f67c python`call_function + 236
    frame #30: 0x000000010015d175 python`_PyEval_EvalFrameDefault + 46837
    frame #31: 0x00000001001508c9 python`_PyEval_EvalCodeWithName + 425
    frame #32: 0x00000001001a955c python`PyRun_FileExFlags + 252
    frame #33: 0x00000001001a8a34 python`PyRun_SimpleFileExFlags + 372
    frame #34: 0x00000001001cf7c6 python`Py_Main + 3734
    frame #35: 0x0000000100000f59 python`main + 313
    frame #36: 0x00007fff758e4015 libdyld.dylib`start + 1
    frame #37: 0x00007fff758e4015 libdyld.dylib`start + 1
```

jmwright, 2019-01-19:
@bernhard-42 If you vary your fillet radius between 0.5 and 0.9 does it help at all?

Also, what happens if you comment the print statements out and move the fillet call up to the box2 line? I ask because I get weird results in CQ-editor when including the print statements.

bernhard-42, 2019-01-19:
Switched to 

```
import cadquery as cq

b = cq.Workplane('XY')
box1 = b.box(10,20,30).edges(">X").chamfer(0.1)
box2 = b.transformed(offset=cq.Vector(0, 1.5, 0.8)).box(3,2,1).edges(">Z").fillet(0.5)
```

Same result

bernhard-42, 2019-01-19:
note, I increase the box by factor 10 to avoid fillet issues with too large values compared to the solid it is applied to

bernhard-42, 2019-01-19:
I mean, I should have done

```
import cadquery as cq

b = cq.Workplane('XY')
box1 = b.box(10,20,30).edges(">X").chamfer(0.1)
box2 = b.transformed(offset=cq.Vector(0, 1.5, 0.8)).box(30,20,10).edges(">Z").fillet(0.5)
```

However still fails.

A few frames, copied after Ctrl-C and then continued

```
  * frame #0: 0x00007fff75aa499d libsystem_malloc.dylib`tiny_free_no_lock + 1149
    frame #1: 0x00007fff75aa5256 libsystem_malloc.dylib`free_tiny + 628
    frame #2: 0x000000010504aaf9 libTKGeomBase.11.dylib`AppCont_LeastSquare::FixSingleBorderPoint(AppCont_Function const&, double, double, double, NCollection_Array1<gp_Pnt2d>&, NCollection_Array1<gp_Pnt>&) + 1657
```

```
  * frame #0: 0x00007fff75a509ce libsystem_m.dylib`___lldb_unnamed_symbol107$$libsystem_m.dylib + 782
    frame #1: 0x000000010504a7ec libTKGeomBase.11.dylib`AppCont_LeastSquare::FixSingleBorderPoint(AppCont_Function const&, double, double, double, NCollection_Array1<gp_Pnt2d>&, NCollection_Array1<gp_Pnt>&) + 876
    frame #2: 0x000000010504bfa8 libTKGeomBase.11.dylib`AppCont_LeastSquare::AppCont_LeastSquare(AppCont_Function const&, double, double, AppParCurves_Constraint, AppParCurves_Constraint, int, int) + 5096
```

```
  * frame #0: 0x00000001056352e3 libTKMath.11.dylib`ElCLib::EllipseD1(double, gp_Ax2 const&, double, double, gp_Pnt&, gp_Vec&) + 115
    frame #1: 0x0000000104fe6284 libTKGeomBase.11.dylib`ProjLib_Function::D1(double, NCollection_Array1<gp_Vec2d>&, NCollection_Array1<gp_Vec>&) const + 228
    frame #2: 0x000000010504afdc libTKGeomBase.11.dylib`AppCont_LeastSquare::AppCont_LeastSquare(AppCont_Function const&, double, double, AppParCurves_Constraint, AppParCurves_Constraint, int, int) + 1052
```

```
  * frame #0: 0x00000001056352e3 libTKMath.11.dylib`ElCLib::EllipseD1(double, gp_Ax2 const&, double, double, gp_Pnt&, gp_Vec&) + 115
    frame #1: 0x0000000104fe6284 libTKGeomBase.11.dylib`ProjLib_Function::D1(double, NCollection_Array1<gp_Vec2d>&, NCollection_Array1<gp_Vec>&) const + 228
    frame #2: 0x000000010504afdc libTKGeomBase.11.dylib`AppCont_LeastSquare::AppCont_LeastSquare(AppCont_Function const&, double, double, AppParCurves_Constraint, AppParCurves_Constraint, int, int) + 1052
```

bernhard-42, 2019-01-19:
This is the minimum failing example

```
import cadquery as cq
box = cq.Workplane('XY').box(1,2,3).edges(">Z").fillet(0.3)
```

since 

```
import cadquery as cq
box = cq.Workplane('XY').box(1,2,3).edges(">Z") 
```

Note: the fillet version works as expected with the conda forge cadquery version

[image]

adam-urbanczyk, 2019-01-19:
@bernhard-42  thanks, clear how to reproduce. For completeness could you paste a full backtrace (just one) from your lldb experiments?

bernhard-42, 2019-01-19:
what different to the callstack above created with `bt`would you need?

adam-urbanczyk, 2019-01-19:
@bernhard-42 it is exactly what I meant, but would be good to have all frames to have a better picture.

BTW: I requested free MacOS workers from CircleCI. If I understand correctly they allow ssh access for debugging builds.

adam-urbanczyk, 2019-02-14:
@bernhard-42 I think we sorted it out. Please look at PR #57 . The secret was different compilation options for `oce`. You could try it already or wait until it is merged (still want to get an OK from @dcowden ).

adam-urbanczyk, 2019-02-15:
#57 is now merged to master. All CI tests pass including those on Mac.

closed 2019-02-15 by adam-urbanczyk