CadQuery/cadquery #79: pythonocc and high accuracy models?

fragmuffin, 2019-01-05:
This is a followup from issues discussed in: #74

So, ideally, models created with `cadquery` can be used for precision (or at least semi-precise) construction with CNC such as 3d printing.

However, I'm having a lot of trouble with pythonocc and high-precision tessellation.

**Example**
The following creates a cylinder at the origin, 1 unit radius, 1 unit high... let's just say they're millimetres.

```python
>>> import cadquery
>>> from time import time
>>> cyl = cadquery.Workplane('XY').circle(1).extrude(1)
>>> tes = cyl.val().tessellate(0.1)  # works ok
>>> tes = cyl.val().tessellate(0.01)  # CPU core @ 100%, docker container needed to be killed
```

For something like a CNC file, I would expect something closer to 1/1000th of a mm accuracy.

**Performance**
```python
>>> (t1, tes, t2) = (time(), cyl.val().tessellate(0.1), time())
>>> len(tes[0])  # count vertices
1006
>>> t2 - t1
0.2510347366333008
```

Also, 250ms for 1 cylinder seems a bit slow, or am I crazy?
That's on an i7 (4-core) running inside a docker container, a bit crude for a performance test, but still.

**`pythonocc` documentation?**
Also, what is a good resource for pythonocc API documentation?
I thought maybe I'm misunderstanding how to use it.

The one linked from http://www.pythonocc.org seems to be [lacking](https://cdn.rawgit.com/tpaviot/pythonocc-core/804f7f3/doc/apidoc/0.18.1/OCC.Visualization.html#OCC.Visualization.Tesselator.Compute)

adam-urbanczyk, 2019-01-05:
@fragmuffin  the tesselator you are using is intended for visualization (judging by the submodule name).

I'd try `exportStl` first, but no promises about quality/speed. Comparing what we have with FreeCAD (https://github.com/FreeCAD/FreeCAD/blob/e02695b91fc8aa19e64594c12386104368809ebc/src/Mod/Part/App/TopoShape.cpp#L815) I see they use absolute deflection. Maybe something to try too?

In general if you want precision for CNC work you should use STEP format which is lossless (unlike STL).

Regarding documentation I think this https://www.opencascade.com/doc/occt-6.9.1/refman/html/index.html is the best place to look. The tesselator you are referring to is not a part of OCC kernel though. This part is of interest I think
https://www.opencascade.com/doc/occt-6.9.1/refman/html/class_b_rep_mesh___incremental_mesh.html#a2aa0a7faf56ed1c1a506af8fadb105e3

fragmuffin, 2019-01-06:
@adam-urbanczyk 

> I'd try exportStl first, but no promises about quality/speed.

OK! thanks... I'll give that a go when I get a chance and post what I find back here.

> In general if you want precision for CNC work you should use STEP format which is lossless (unlike STL).

Good point, CNC may be a bad example 😉 .
But I still imagine there will be valid use-cases, perhaps just importing a STEP file to output an STL with a chosen precision.... but yes, I'll try `exportStl`.

> Regarding documentation I think this https://www.opencascade.com/doc/occt-6.9.1/refman/html/index.html is the best place to look.

Thanks :+1:,

adam-urbanczyk, 2019-06-07:
Can I close this issue @fragmuffin ?

adam-urbanczyk, 2019-06-13:
If this is still an issue feel free to reopen @fragmuffin

closed 2019-06-13 by adam-urbanczyk