CadQuery/cadquery #80: one surface with different color with pythonocc

liuxin2322, 2019-01-06:
just like this.  how to realize it ?
[image]

jmwright, 2019-01-06:
See #60

While CadQuery doesn't have good support for surfaces yet, there might be lower level PythonOCC functionality that you can access to do what you want. You can mix PythonOCC syntax with CadQuery syntax since they're both based on Python.

adam-urbanczyk, 2019-02-04:
I think you need to experiment with this class from PythonOCC: https://www.opencascade.com/doc/occt-6.9.0/refman/html/class_a_i_s___textured_shape.html . Unfortunately I have no experience with it. Please also note that this is visualization related class (i.e. `AIS_*`).  

I will close this issue for now.

closed 2019-02-04 by adam-urbanczyk