CadQuery/cadquery #83: Conda Install of Base CadQuery Installs PyQt and Qt

jmwright, 2019-01-16:
When running the conda installation line given in the readme, PyQt and Qt are installed. If someone just wants the core library, they shouldn't need PyQt/Qt and their support libraries should they? They only need those if they're going to run CQ-editor, correct?

adam-urbanczyk, 2019-01-17:
Thanks, this is pythonocc related https://github.com/tpaviot/pythonocc-core/issues/538

I am building custom packages now via azure and this repo: https://github.com/CadQuery/conda-packages . Will edit the recipe there.

adam-urbanczyk, 2019-06-07:
PyQT / Qt is not longer required by our build of pythonOCC:

https://raw.githubusercontent.com/CadQuery/conda-packages/master/pythonocc/meta.yaml

closed 2019-06-07 by adam-urbanczyk