CadQuery/cadquery #100: Time for new release

adam-urbanczyk, 2019-02-16:
I think it would be good to make a new release. Which number should it be 2.0, 1.3 or something completely different?

I personally would not call what we have so far CQ2.0, but I am very much interested in what everyone thinks @dcowden @jmwright .

adam-urbanczyk, 2019-02-16:
It would be also good to have this closed before #12 is finalized

jmwright, 2019-02-17:
@adam-urbanczyk I think the logic behind the 2.0 version number was tied to semantic versioning where changing that first number signifies a breaking change in the API. While the user-facing API is mostly unchanged, the layer below that has changed significantly, and the installation process is drastically different than CQ 1.0. That's the logic anyway.

zignig, 2019-02-17:
I think sticking to semver is a  good idea. As this is a different basis (occ, rather than FreeCad) I think a fresh  number is warranted.

2.0.0-alpha

I am keen to help transmogrify my cqparts collection across to the new structure.

adam-urbanczyk, 2019-02-21:
Alright, 2.0.0 it is.

fragmuffin, 2019-02-23:
I'm also considering how to handle this within `cqparts`.

I've outlined a version structure for `cqparts` in the [`__init__.py`](https://github.com/cqparts/cqparts/blob/c8f1c7ad441435f34322ab8962aefd41938b316e/src/cqparts/__init__.py#L17-L24) file.
```
# =========================== Package Information ===========================
# Version Planning:
#   0.1.x               - Development Status :: 2 - Pre-Alpha
#   0.2.x               - Development Status :: 3 - Alpha
#   0.3.x               - Development Status :: 4 - Beta
#   1.x                 - Development Status :: 5 - Production/Stable
#   <any above>.y       - developments on that version (pre-release)
#   <any above>*.dev*   - development release (intended purely to test deployment)
```
`cqparts` is overdue for a release as well (`cqparts==0.2.1` at this time)

**My Plan**
* `v0.2.2` - recent updates to freecad supported lib
* `v0.2.3` - last release to support freecad
* \>= `v0.2.4` - pythonocc release

What do you think?

jmwright, 2019-02-23:
@fragmuffin If you're going by semantic versioning, the bugfix bump from 0.2.3 to 0.2.4 for PythonOCC isn't really appropriate is it?

dcowden, 2019-02-23:
+1 I think semver is my vote. With x.Y. Z, id expect z changes to be minor and both forwards and backwards api compatible
Y changes are backward but not forward compatible and add new features 
X changes might not necessarily be backwards compatible and have major newness. 

In that sense, for cq I'd Say 2.0 is appropriate because I'm not sure it's totally backwards api compatible. even if it is, relative to the fact that it's not freecad based any longer it is not compatible for folks who had freecad code.

So I'd probably vote for cq 2.0.0

adam-urbanczyk, 2019-02-24:
I created a CQ2.0RC project (i.e. things that need to be in place before we can start talking about a 2.0 release. Please take a look and let me know what do you think about the scope

fragmuffin, 2019-03-04:
@jmwright I haven't been considering a semantic version scheme because I still consider `cqparts` to be at alpha maturity.
But you're right; semantic portability is still relevant to the users.

adam-urbanczyk, 2019-06-11:
@dcowden @jmwright I think everything is in place for a release. What do you think? Are there any critical issues that should be closed?

dcowden, 2019-06-11:
@adam-urbanczyk i'm totally behind so i am going to claim baggage here completely.

jmwright, 2019-06-11:
@adam-urbanczyk I think that as long as it works on all three major OSes (and I think it does at this point), I'm good with doing a release.

jmwright, 2021-02-05:
Looks like we forgot to close this issue.

closed 2021-02-05 by jmwright