CadQuery/cadquery #102: MacOS install issue?

osterwood, 2019-02-20:
When running this
```
conda install -c pythonocc -c oce -c conda-forge -c dlr-sc -c CadQuery cadquery-occ python=3.6
```
I get this error message
```
Collecting package metadata: done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - cadquery-occ

Current channels:

  - https://conda.anaconda.org/pythonocc/osx-64
  - https://conda.anaconda.org/pythonocc/noarch
  - https://conda.anaconda.org/oce/osx-64
  - https://conda.anaconda.org/oce/noarch
  - https://conda.anaconda.org/conda-forge/osx-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://conda.anaconda.org/dlr-sc/osx-64
  - https://conda.anaconda.org/dlr-sc/noarch
  - https://conda.anaconda.org/CadQuery/osx-64
  - https://conda.anaconda.org/CadQuery/noarch
  - https://repo.anaconda.com/pkgs/main/osx-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/free/osx-64
  - https://repo.anaconda.com/pkgs/free/noarch
  - https://repo.anaconda.com/pkgs/r/osx-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

I was able to install pythonocc-core successfully with 
```
conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1 python=3.6
```
and also tried installing cadquery from the repo with the normal `python setup.py install` Running anything results in:

```
>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/chris/Documents/github/cadquery/cadquery/cadquery/__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix
  File "/Users/chris/Documents/github/cadquery/cadquery/cadquery/occ_impl/geom.py", line 4, in <module>
    from OCC.Core.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp, gp_XYZ
ModuleNotFoundError: No module named 'OCC.Core'
```

Any ideas?  I'm not the most familiar with conda packaging.

adam-urbanczyk, 2019-02-21:
Currently the readme info is outdated (we are working on it here #99). Could you try:
```python
conda install -c conda-forge -c cadquery pythonocc-core=0.18.2 pyparsing python=3.6
pip install git+https://github.com/CadQuery/cadquery.git
```

adam-urbanczyk, 2019-02-24:
@osterwood did it work?

osterwood, 2019-02-24:
I’ve been out of town, so unable to try.  I’ll let you know next weekend.  Thanks for the suggestion

bernhard-42, 2019-03-03:
I gave it a try and it works. 

However, it replaces my anaconca python 3.6.8 with 3.6.7:
`python:          3.6.8-haf84260_0                  --> 3.6.7-h8dc6b48_1004   conda-forge`
Where does this dependency come from?

osterwood, 2019-03-04:
Yep - that works for me.  I get a few deprecation warnings, but cadquery imports fine.  Thanks so much.

```
/Users/chris/anaconda3/envs/cq/lib/python3.6/site-packages/OCC/LocOpe.py:3: DeprecationWarning: OCC.LocOpe is deprecated since pythonocc-0.18.2. Use OCC.Core.LocOpe
  warnings.warn("OCC.LocOpe is deprecated since pythonocc-0.18.2. Use OCC.Core.LocOpe", DeprecationWarning)
/Users/chris/anaconda3/envs/cq/lib/python3.6/site-packages/OCC/BRepCheck.py:3: DeprecationWarning: OCC.BRepCheck is deprecated since pythonocc-0.18.2. Use OCC.Core.BRepCheck
  warnings.warn("OCC.BRepCheck is deprecated since pythonocc-0.18.2. Use OCC.Core.BRepCheck", DeprecationWarning)
```

closed 2019-03-04 by osterwood

mrwonko, 2019-05-16:
When I try to
```bash
conda install -c conda-forge -c cadquery pythonocc-core=0.18.2 pyparsing python=3.6
```
I get

```
ERROR conda.core.link:_execute(568): An error occurred while uninstalling package 'defaults::python.app-2-py37_9'.
PermissionError(1, 'Operation not permitted')
Attempting to roll back.

Rolling back transaction: done

[Errno 1] Operation not permitted: '/miniconda3/python.app/Contents/PkgInfo' -> '/miniconda3/python.app/Contents/PkgInfo.c~'
```

adam-urbanczyk, 2019-05-16:
@mrwonko could you update conda and try again?