CadQuery/cadquery #118: Polar array support with changing orientation

adam-urbanczyk, 2019-03-22:
Currently there is no easy way to achieve the right result:

[image]

Wheras it is trival to get to the left one:
```python
r1 = cq.Workplane("XY" )\
    .box(10, 10, 1).faces(">Z")\
    .polarArray(3.5,0,360,6).rect(1.6,1.6).cutThruAll()
```

I think the right scenario occurs quite frequently, so it would make sense to support it.

Related to:
https://github.com/CadQuery/CQ-editor/issues/31

adam-urbanczyk, 2019-06-07:
I was thinking about how to implement it this and came up with the following design

1. extend `geom.Vector` so that it allows to attach a transform object to itself
2. refactor `polarArray` to create `Vectors` with proper rotation transforms attached
3. refactor `eachpoint` to apply the attached transform if it exists

Any thoughts?

fapbrandt, 2020-01-03:
Hi. Why not in 2.0RC2?

jmwright, 2020-01-03:
@fapbrandt 2.0RC2 is more about finalizing a stable version of CadQuery that runs closer to the OCC CAD kernel than FreeCAD allowed us to. We accept PRs for new features like this for the 2.0 release candidates, but the core development team probably won't have time to get it done before version 2.1.

fapbrandt, 2020-01-03:
Thanks for answer, jmwright. I personally will check cq.py to see what I can do in my own code. See you.

jmwright, 2020-01-03:
@fapbrandt You can probably implement this for use in your own scripts using plugins.

https://cadquery.readthedocs.io/en/latest/extending.html#cadqueryexample-plugins

PR #395 (2020-07-09): polarArray with object rotation
This will resolve #118 

[image]

```python
import cadquery as cq 
b = cq.Workplane().box(10,10,1).faces('>Z').workplane()

e = b.polarArray(3.2,0,360,8,rotate=False).slot2D(3,1,90).cutThruAll()
show_object(e)

e = b.polarArray(3.2,0,360,8,rotate=True).slot2D(3,1,90).cutThruAll()
show_object(e.translate((12,0,0)))
```
closed 2020-07-09 by jmwright