CadQuery/cadquery #119: Exception when adding fillet to part

rowanG077, 2019-03-25:
When adding a fillet to the part an exception is throw.

Example code:

```
from __future__ import division
from cadquery import *
import math

# monkey patch cadquery to allow storing
# and loading of workplanes
def store_workplane(self):
    return self.plane

def load_workplane(plane):
    wp = cq.Workplane("XY")
    wp.plane = plane
    return wp

cq.Workplane.store_workplane = store_workplane
cq.load_workplane = load_workplane

feet_thickness = 4.75
feet_side_length = 12.70
feet_angle = math.radians(80)

# cosine law
feet_diagonal1 = math.sqrt(2 * feet_side_length ** 2 - 2 * math.cos(feet_angle) * feet_side_length ** 2)
feet_diagonal2 = math.sqrt(2 * feet_side_length ** 2 - 2 * math.cos(math.pi - feet_angle) * feet_side_length ** 2)
feet_triangle_height = (math.sqrt(3) / 2) * feet_diagonal1
feet_fillet = 0.8

support_attach_base_thickness = 20
three_point_triangle_dia = 160

# add 2 mm for the glue
three_point_triangle_beam_length = 20 + 100

three_point = cq.Workplane("XZ")\
    .transformed(
        rotate=(90, 0, 90)
    )\
    .polygon(3, three_point_triangle_dia)\
    .extrude(support_attach_base_thickness)

# create feet
def create_feet_wire(self, l, angle):
    offset_y  = l * math.cos(angle / 2)
    offset_x  = l * math.sin(angle / 2)

    return self\
        .line(offset_x, offset_y)\
        .line(-offset_x, offset_y)\
        .line(-offset_x, -offset_y)\
        .close()

def create_feet_beam(self, length):
    cutter = self\
        .transformed(
            offset = (
                -25,
                three_point_triangle_dia/2 - feet_triangle_height,
                0
            )
        )\
        .rect(50, 50, centered = False)\
        .extrude(-50, combine = False)
    
    foot = self\
        .transformed(
            offset = (
                0,
                three_point_triangle_dia/2 - feet_diagonal2/2 - feet_triangle_height,
                0
            )
        )\
        .create_feet_wire(feet_side_length, feet_angle)\
        .extrude(-length, combine = False)
    
    return (cutter, foot)

cq.Workplane.create_feet_wire = create_feet_wire
cq.Workplane.create_feet_beam = create_feet_beam

three_point_workplane = three_point\
    .faces(">X").workplane().store_workplane()

for i in range(0, 3):
    (cutter, foot) = cq.load_workplane(three_point_workplane)\
       .transformed(rotate = (i * 120, 0, 0))\
       .create_feet_beam(three_point_triangle_beam_length)

    three_point = three_point.cut(cutter).union(foot)

three_point = three_point.edges("|X").fillet(1)
show_object(three_point)
```

On the second the last line I add a fillet and it throws an exception. If you replace filllet with sphere you can see that I selected the correct edges. However when I add a fillet to every single edge(without filtering on edges.) on the part it works fine.

jmwright, 2019-03-26:
@rowanG077 Does it throw the same error if you use a chamfer instead of a fillet?

rowanG077, 2019-03-26:
@jmwright Yes it fails with chamfer as well with the same exception.

adam-urbanczyk, 2019-03-26:
@rowanG077 I modified the last part:
 ```python
three_point = three_point.edges("|X").edges("not >X").fillet(feet_fillet)
```
and this is what I get:
[image]

The additional small edge causes this issue. I think you need to split the fillet operation into two. I see the same  with FreeCAD so it seems to be a limitation of the CAD kernel (OCCT).

rowanG077, 2019-03-26:
Yeah it's a problem in OCC. It's still weird because I can still ad a fillet to that small edge if I fillet in two steps.

Thanks for your help!

adam-urbanczyk, 2019-03-27:
I think that if both edges are selected resulting blend is supposed to be continuous (see below). For the geometry in question it is not easy to achieve. 

[image]

adam-urbanczyk, 2019-03-29:
@rowanG077 I want to close this. Do you see any further issues with fillets for your model?

rowanG077, 2019-03-29:
Yes can be closed! Thank you.

closed 2019-03-29 by adam-urbanczyk

jrobichaud, 2020-02-21:
This issue is easy to reproduce when "filleted" faces touch each others.

It can be reproduced this way:

Not working:
```python
	cq.Workplane("front").box(1, 1, 1).edges("+Z").fillet(0.5)
```

Working:
```python
	cq.Workplane("front").box(1, 1, 1).edges("+Z").fillet(0.4999999)
```