CadQuery/cadquery #120: edges() is broken with Key Error: 7

bernhard-42, 2019-04-12:
For complex shapes one can get
```

/opt/miniconda/envs/pycq/lib/python3.6/site-packages/cadquery/occ_impl/shapes.py in geomType(self)
    298             return tr
    299         else:
--> 300             return geom_LUT_EDGE_FACE[tr(self.wrapped).GetType()]
    301 
    302     def isType(self, obj, strType):  # TODO why here?

KeyError: 7
```

**Issue 1**

Looking into it, I found `geom_LUT_EDGE_FACE` being an interesting mixture of edge and face constants:

```
geom_LUT_EDGE_FACE = \
    {ga.GeomAbs_Arc: 'ARC',
     ga.GeomAbs_Circle: 'CIRCLE',
     ga.GeomAbs_Line: 'LINE',
     ga.GeomAbs_BSplineCurve: 'SPLINE',  # BSpline or Bezier?
     ga.GeomAbs_Plane: 'PLANE',
     ga.GeomAbs_Sphere: 'SPHERE',
     ga.GeomAbs_Cone: 'CONE',
     }
```

leading to

```
import cadquery.occ_impl.shapes as s
s.geom_LUT_EDGE_FACE

{0: 'PLANE', 1: 'CIRCLE', 6: 'SPLINE', 3: 'SPHERE', 2: 'CONE'}
```

Reason being that the constants of edges and faces seem to overlap in occ:

```
import OCC.Core.GeomAbs as ga
print(
    ga.GeomAbs_Arc,
    ga.GeomAbs_Circle,
    ga.GeomAbs_Line,
    ga.GeomAbs_BSplineCurve,
    ga.GeomAbs_Plane,
    ga.GeomAbs_Sphere,
    ga.GeomAbs_Cone,
)

0 1 0 6 0 3 2
```

So 'ARC' == 'LINE' == 'PLANE'

I find it especially strange that `GeomAbs_Arc` and `GeomAbs_Line` overlap.

**Issue 2**
Now
```
print(ga.GeomAbs_OtherCurve)

7
```

Brute force setting 

```
import cadquery.occ_impl.shapes
cadquery.occ_impl.shapes.geom_LUT_EDGE_FACE[7] = "OTHER"
```

makes the `edges()` call work, however, no idea of the side effects.

And due to the overlapping indexes, this part of the code might need some love.

adam-urbanczyk, 2019-04-12:
@bernhard-42 thanks for this in-depth bug report! Could you also paste a code example that triggers this error?

adam-urbanczyk, 2019-04-12:
`GeomAbs_Arc` is from a different enum:
https://www.opencascade.com/doc/occt-6.9.0/refman/html/_geom_abs___join_type_8hxx.html#ad03b8c9363e1c78dc022d2f253273c61

bernhard-42, 2019-04-13:
Here is a sample that breaks with KeyError: 7

```
import cadquery as cq

diam = 5.2
thickness = 2
width = 32
length = 60

xy = cq.Workplane("YX")
base = (xy
    .vLineTo(-length/2, forConstruction=True)
    .hLineTo(-width/2)
    .vLineTo(length / 6)
    .threePointArc((-width*3/8, length*3/8), (0, length/2))
    .mirrorY()
    .extrude(thickness)
    .translate((0, 0, thickness/2))
    .edges("|Z and (not >Y) and (not <Y)").fillet(0.7)
    .edges(">Z or <Z").fillet(0.75)
)

base.edges("not |X")
```

my cadquery is installed via  (see https://github.com/CadQuery/cadquery/issues/102)

```
conda install -c conda-forge -c cadquery pythonocc-core=0.18.2 pyparsing python=3.6
pip install git+https://github.com/CadQuery/cadquery.git
```

with `Python 3.6.7 | packaged by conda-forge | (default, Feb 28 2019, 02:16:08)`  on macos

bernhard-42, 2019-04-13:
When I change the radius of the fillet in the line marked with ISSUE to 0.8, the KeyError does not appear. However, for both values the objects renders well.

Btw. the reason why I came across this is that I currently re-implement my jupyter extension to use pythreejs instead of x3dom and added a feature to visualise selected edges, since I always find it difficult to ensure that my selections really select what I intended. So visually seeing the selected edges on the object is really helpful.

PR #121 (2019-05-08): Fix face and edge geometry LUTs
WIP - do not merge. This PR will resolve #120
closed 2019-05-08 by adam-urbanczyk