CadQuery/cadquery #124: Shell.makeShell error

rusc444, 2019-05-01:
I'm trying to build polygons in cadquery from vertices/faces definitions found here:

[http://dmccooey.com/polyhedra/](url)

A code example for the tetrahedron:

```
import cadquery as cq
import math

c0 = math.sqrt(2)/4
vertices = [[c0, -c0,  c0], [c0,  c0, -c0], [-c0,  c0,  c0], [-c0, -c0, -c0]]
faces = [[0, 1, 2], [1, 0, 3], [2, 3, 0], [3, 2, 1]]

l_faces = []
for aface in faces:
    aface.append(aface[0])
    l_lines = []
    for avertex in range(len(aface)-1):
        l_lines.append(cq.Edge.makeLine(cq.Vector(*vertices[aface[avertex]]),
                       cq.Vector(*vertices[aface[avertex+1]])))
    wire = cq.Wire.combine(l_lines)
    l_faces.append(cq.Face.makeFromWires(wire))
ashell = cq.Shell.makeShell(l_faces)
```

but I get an error in makeShell method:

> File "C:\Miniconda3\envs\cq\lib\site-packages\cadquery\occ_impl\shapes.py", line 1024, in makeShell
    shell_builder.Add(face.wrapped)
TypeError: TopoDS_Builder_Add expected 3 arguments, got 2

As it seems, the correct method call has to be 

`shell_builder.Add(shell_wrapped, face.wrapped)`

instead of

`shell_builder.Add(face.wrapped)`

This modified code works for me and the result seems to be a cadquery closed shell.

The next problem is that I don't see any way to convert this shell to a cadquery solid to work with. Is this possible ?.

Thanks

adam-urbanczyk, 2019-05-01:
Thanks for reporting this @rusc444 ! Since you already found a solution, would you consider submitting a PR?

Regarding shell to solid conversion: It should be possible with the `BRepBuilderAPI_MakeSolid` OCC class. If you feel like it, you could try implement a `makeSolid` method (similar to `makeShell`)

rusc444, 2019-05-03:
Thank you for the quick response.

I am not a professional programmer or a specialized CAD user. I just like to program in Python and use a 3d printer in hobby projects and cadquery is a awesome tool to combine the two interests. I do not have much knowledge of the procedures involving code development in GitHub and I think that is better if you manage the PR.

Regarding the shell to solid conversion, thanks for the suggestion. I'll try to do it using PythonOCC scripting.

adam-urbanczyk, 2019-05-04:
No worries @rusc444 , I'll make a PR based on your report then and try to include the `makeSolid` method. Thanks again for your detailed bug report!

adam-urbanczyk, 2019-05-20:
Closed in #128

closed 2019-05-20 by adam-urbanczyk