CadQuery/cadquery #127: Guard lineTo to avoid malloc errors

bernhard-42, 2019-05-11:
Unfortunately, cadquery sometimes dies with malloc errors. One source for these errors is, when you `lineTo` a point that is by accident the same as the last Point. How about  guarding `Workplane.lineTo`? One option could be:

```
def lineTo(self, x, y, forConstruction=False):
    """
    Make a line from the current point to the provided point

    :param float x: the x point, in workplane plane coordinates
    :param float y: the y point, in workplane plane coordinates
    :return: the Workplane object with the current point at the end of the new line

    see :py:meth:`line` if you want to use relative dimensions to make a line instead.
    """
    startPoint = self._findFromPoint(False)

    endPoint = self.plane.toWorldCoords((x, y))

    if startPoint.sub(endPoint).Length > self.ctx.tolerance:
        p = Edge.makeLine(startPoint, endPoint)

        if not forConstruction:
            self._addPendingEdge(p)

        return self.newObject([p])
    else:
        return self
```

adam-urbanczyk, 2019-05-16:
@bernhard-42 I am not sure if I follow. Could you provide an example script to trigger this behavior? A backtrace from a debugger would be also appreciated.

bernhard-42, 2019-05-16:
Let me give you three examples:

1) lineTo, when start and end point are the same

```python
Python 3.6.7 | packaged by conda-forge | (default, Feb 28 2019, 02:16:08)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.3.0 - An enhanced Interactive Python. Type '?' for help.

In [1]: import cadquery

In [2]: cadquery.Workplane("XY").lineTo(0,0)
python(14646,0x7fff957d4380) malloc: *** error for object 0x7fc21f15e2f8: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6
```
2) fillet (or chamfer) when radius is too large

```python
Python 3.6.7 | packaged by conda-forge | (default, Feb 28 2019, 02:16:08)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.3.0 - An enhanced Interactive Python. Type '?' for help.

In [1]: import cadquery

In [2]: cadquery.Workplane("XY").box(1,1,1).edges(">Z").fillet(1)
python(14721,0x7fff957d4380) malloc: *** error for object 0x7fc063087488: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6
```

3) threePointArc when the three points are on one line

```python
Python 3.6.7 | packaged by conda-forge | (default, Feb 28 2019, 02:16:08)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.3.0 - An enhanced Interactive Python. Type '?' for help.

In [1]: import cadquery

In [2]: cadquery.Workplane("XY").threePointArc((1,1), (2,2))
python(14734,0x7fff957d4380) malloc: *** error for object 0x7fd81330a278: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6
```

All three example are more or less sense less, how can happen by accident. The annoying part is that (at least on macos) the python interpreter crashes. So no chance in catching the issue.

adam-urbanczyk, 2019-05-16:
On linux I get normal exception: `StdFail_NotDone` or `StdFail_NotDone BRep_API`.

Could you run it with lldb or gdb and display the backtrace? I wonder wheter this error happens in pyhonocc bindings or in oce proper.

bernhard-42, 2019-05-16:
yeah, just tried on my linux machine, I also get a python exception.

Not that I understand anything of the below ;-) is this what you are interested in?

```
 lldb
(lldb) file /opt/miniconda/envs/pycq/bin/python
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/opt/miniconda/lib/python2.7/copy.py", line 52, in <module>
    import weakref
  File "/opt/miniconda/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
Current executable set to '/opt/miniconda/envs/pycq/bin/python' (x86_64).
(lldb) process launch
Process 14915 launched: '/opt/miniconda/envs/pycq/bin/python' (x86_64)
Python 3.6.7 | packaged by conda-forge | (default, Feb 28 2019, 02:16:08)
[GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery
>>> cadquery.Workplane("XY").box(1,1,1).edges(">Z").fillet(1)
python(14915,0x7fff957d4380) malloc: *** error for object 0x1168510d8: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Process 14915 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff5d209b66 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff5d209b66 <+10>: jae    0x7fff5d209b70            ; <+20>
    0x7fff5d209b68 <+12>: movq   %rax, %rdi
    0x7fff5d209b6b <+15>: jmp    0x7fff5d200ae9            ; cerror_nocancel
    0x7fff5d209b70 <+20>: retq
Target 0: (python) stopped.
(lldb) thread backtrace all
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff5d209b66 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff5d3d4080 libsystem_pthread.dylib`pthread_kill + 333
    frame #2: 0x00007fff5d1651ae libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff5d2638a6 libsystem_malloc.dylib`free + 521
    frame #4: 0x00000001070c2d69 libc++abi.1.dylib`__cxa_end_catch + 137
    frame #5: 0x000000010970e867 _BRepBuilderAPI.so`_wrap_BRepBuilderAPI_MakeShape_Shape(_object*, _object*) + 2311
    frame #6: 0x000000010008b5b2 python`_PyCFunction_FastCallDict + 322
    frame #7: 0x0000000100011e81 python`_PyObject_FastCallKeywords + 385
    frame #8: 0x0000000100162688 python`call_function + 392
    frame #9: 0x0000000100160111 python`_PyEval_EvalFrameDefault + 46817
    frame #10: 0x000000010016294c python`fast_function + 188
    frame #11: 0x00000001001625ec python`call_function + 236
    frame #12: 0x0000000100160111 python`_PyEval_EvalFrameDefault + 46817
    frame #13: 0x000000010016294c python`fast_function + 188
    frame #14: 0x00000001001625ec python`call_function + 236
    frame #15: 0x0000000100160111 python`_PyEval_EvalFrameDefault + 46817
    frame #16: 0x0000000100153869 python`_PyEval_EvalCodeWithName + 425
    frame #17: 0x00000001001ae636 python`PyRun_InteractiveOneObjectEx + 1126
    frame #18: 0x00000001001ab2f1 python`PyRun_InteractiveLoopFlags + 433
    frame #19: 0x00000001001ab11b python`PyRun_AnyFileExFlags + 139
    frame #20: 0x00000001001d25b0 python`Py_Main + 4720
    frame #21: 0x0000000100001d19 python`main + 313
    frame #22: 0x00007fff5d0b9015 libdyld.dylib`start + 1
(lldb)
```

adam-urbanczyk, 2019-05-16:
Yes, spot on. It looks like the error happens in the wrapper (pythonOCC):
```
frame #5: 0x000000010970e867 _BRepBuilderAPI.so`_wrap_BRepBuilderAPI_MakeShape_Shape(_object*, _object*) + 2311
```

bernhard-42, 2019-05-16:
fyi, the circle example also fails in the wrapper

```
>>> cadquery.Workplane("XY").threePointArc((1,1), (2,2))
python(14979,0x7fff957d4380) malloc: *** error for object 0x110acad78: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Process 14979 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff5d209b66 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff5d209b66 <+10>: jae    0x7fff5d209b70            ; <+20>
    0x7fff5d209b68 <+12>: movq   %rax, %rdi
    0x7fff5d209b6b <+15>: jmp    0x7fff5d200ae9            ; cerror_nocancel
    0x7fff5d209b70 <+20>: retq
Target 0: (python) stopped.
(lldb) thread backtrace all
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff5d209b66 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff5d3d4080 libsystem_pthread.dylib`pthread_kill + 333
    frame #2: 0x00007fff5d1651ae libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff5d2638a6 libsystem_malloc.dylib`free + 521
    frame #4: 0x00000001067a0d69 libc++abi.1.dylib`__cxa_end_catch + 137
    frame #5: 0x000000010aaa8580 _GC.so`_wrap_GC_MakeArcOfCircle_Value(_object*, _object*) + 1104
    frame #6: 0x000000010008b5b2 python`_PyCFunction_FastCallDict + 322
    frame #7: 0x0000000100011e81 python`_PyObject_FastCallKeywords + 385
    frame #8: 0x0000000100162688 python`call_function + 392
    frame #9: 0x0000000100160111 python`_PyEval_EvalFrameDefault + 46817
    frame #10: 0x000000010016294c python`fast_function + 188
    frame #11: 0x00000001001625ec python`call_function + 236
    frame #12: 0x0000000100160111 python`_PyEval_EvalFrameDefault + 46817
    frame #13: 0x0000000100153869 python`_PyEval_EvalCodeWithName + 425
    frame #14: 0x00000001001629fa python`fast_function + 362
    frame #15: 0x00000001001625ec python`call_function + 236
    frame #16: 0x0000000100160111 python`_PyEval_EvalFrameDefault + 46817
    frame #17: 0x0000000100153869 python`_PyEval_EvalCodeWithName + 425
    frame #18: 0x00000001001ae636 python`PyRun_InteractiveOneObjectEx + 1126
    frame #19: 0x00000001001ab2f1 python`PyRun_InteractiveLoopFlags + 433
    frame #20: 0x00000001001ab11b python`PyRun_AnyFileExFlags + 139
    frame #21: 0x00000001001d25b0 python`Py_Main + 4720
    frame #22: 0x0000000100001d19 python`main + 313
    frame #23: 0x00007fff5d0b9015 libdyld.dylib`start + 1
(lldb)
```

adam-urbanczyk, 2019-05-17:
Could you try how the following code works with pythonocc-core from the channel of `tpaviot`? If you get the same issue I'd say it is an upstream thing. If not, I probably did something wrong when building.
```python 
from OCC.Core.gp import gp_Pnt
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge

BRepBuilderAPI_MakeEdge(gp_Pnt(0,0,0),gp_Pnt(0,0,0)).Edge()
```

bernhard-42, 2019-05-18:
Environment:

```bash
conda create -n occ python=3.6
conda activate occ
conda install -c pythonocc -c oce -c dlr-sc smesh
conda install -c tpaviot pythonocc-core==0.18.2
```

Result:
```
Python 3.6.8 |Anaconda, Inc.| (default, Dec 29 2018, 19:04:46)
[GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from OCC.Core.gp import gp_Pnt
>>> from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge
>>> BRepBuilderAPI_MakeEdge(gp_Pnt(0,0,0),gp_Pnt(0,0,0)).Edge()
python(22587,0x7fff957d4380) malloc: *** error for object 0x7fe5e972bb88: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6
```

So you're right, the issue is upstream. Will open an issue with pythonocc-core

closed 2019-05-18 by bernhard-42