CadQuery/cadquery #130: The difference between "rotate" under Ubuntu and windows

WTG2017, 2019-05-21:
Hello, thank you for seeing my question. Fortunately, i have the opportunity to use your software in my work. One problem we found today is that “rotate” commands seem to be on different platforms.On ubuntu（conad3+pythonOCC+CQ-Editor）, rotate ((1, 0, 0), (0, 0), 90 * (math. pi / 180)) is the result I expected，On Windows 7（conda3+pythonOCC）, rotate ((1, 0, 0), (0, 0), 0, 90) can achieve the desired results.

jmwright, 2019-05-22:
Hi @WTG2017 

Can you post a more complete example of what you're trying to do?

WTG2017, 2019-05-23:
The same code, in different environments, has different results。
windows use angle：
[image]
ubuntu use angle
[image]
Can be modified to radian under ubuntu, the results are consistent
ubuntu use radian
[image]

adam-urbanczyk, 2019-05-23:
Wow that is super weird @WTG2017, I cannot reproduce it though. Could you pip install latest cadquery version from git: `pip install git+https://github.com/CadQuery/cadquery.git` ?

WTG2017, 2019-05-24:
Before this I have downloaded cadquery source code  from https://github.com/CadQuery/cadquery，and used the command “python setup.py install”to install it, but there has the "rotate problem" issue. Then I uninstalled it, and used the command "pip install git+https://github.com/CadQuery/cadquery.git" to reinstall it, but there has the issue still. Tomorrow I will  install a new version CQ-Editor on ubuntu and trying to find out which step went wrong.

adam-urbanczyk, 2019-05-31:
Any updates @WTG2017 ?

adam-urbanczyk, 2019-06-25:
Closing, feel free to reopen @WTG2017 if this is still relevant.

closed 2019-06-25 by adam-urbanczyk