CadQuery/cadquery #131: Problem with install

yguel, 2019-05-22:
On the main page it is explain how to install cadquery version working with occ:

conda install -c pythonocc -c oce -c conda-forge -c dlr-sc -c CadQuery cadquery-occ

However I get the following error:
VVVVVVVVVVVVVVVVVVVVVVVVVVVVVV
WARNING: The conda.compat module is deprecated and will be removed in a future release.
Collecting package metadata: done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - cadquery-occ

Current channels:

  - https://conda.anaconda.org/pythonocc/linux-64
  - https://conda.anaconda.org/pythonocc/noarch
  - https://conda.anaconda.org/oce/linux-64
  - https://conda.anaconda.org/oce/noarch
  - https://conda.anaconda.org/conda-forge/linux-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://conda.anaconda.org/dlr-sc/linux-64
  - https://conda.anaconda.org/dlr-sc/noarch
  - https://conda.anaconda.org/CadQuery/linux-64
  - https://conda.anaconda.org/CadQuery/noarch
  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/free/linux-64                                                                                                                                                                   
  - https://repo.anaconda.com/pkgs/free/noarch                                                                                                                                                                     
  - https://repo.anaconda.com/pkgs/r/linux-64                                                                                                                                                                      
  - https://repo.anaconda.com/pkgs/r/noarch                                                                                                                                                                        
                                                                                                                                                                                                                   
To search for alternate channels that may provide the conda package you're                                                                                                                                         
looking for, navigate to                                                                                                                                                                                           
                                                                                                                                                                                                                   
    https://anaconda.org                                                                                                                                                                                           
                                                                                                                                                                                                                   
and use the search bar at the top of the page.
/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\/\

I am new to the use of conda, so did I missed something obvious or is there a change that made the install hint obsolete ?

adam-urbanczyk, 2019-05-22:
Thanks for your interest in cadquery @yguel . README is currently outdated (there is a PR in progress to rectify it).

Cadquery can be installed using conda (please use a recent version) like so: 
```
conda install -c cadquery -c conda-forge python=3.6 pythonocc-core=0.18.2 oce=0.18.2 pyparsing
pip install git+https://github.com/CadQuery/cadquery.git
```
Let me know if this works for you.

yguel, 2019-05-24:
Thanks a lot Adam,
It works at the first try. However when trying some examples, the show commands do not work. What would be the changes to cadquery/examples/Ex001_Simple_Block.py that allow to see the result ?

adam-urbanczyk, 2019-05-24:
You need a tool to run those examples in (again outdated README; we are really working on it). My personal advice is to use CQ-editor: https://github.com/CadQuery/CQ-editor:
[image]
Installation instruction of it should work out of the box (for a change).

adam-urbanczyk, 2019-05-29:
@yguel any luck?

yguel, 2019-05-29:
Yes, it is working since today. I had many problems, here is the complete solution for my ubuntu 18.04:

  * Install pyqt5 via the package manager of ubuntu (synaptic for instance) ''python3-pyqt5''.

  * Install all packages related to pyqt5

  * Verify that there is no other install of pyqt5, especially with pip
  >>pip uninstall PyQt5
  >>pip3 uninstall PyQt5

  * verify that ''python'' bash command executes ''/usr/bin/python3'' and not some ''~/local/conda/...''

  * Install [CQ-editor](https://github.com/CadQuery/CQ-editor)
  >>conda env create -f cqgui_env.yml -n cqgui
  >>conda activate cqgui

  * Install cadquery
  >>conda install -c cadquery -c conda-forge python=3.6 pythonocc-core=0.18.2 oce=0.18.2 pyparsing

  * Run CQ-editor
  >>python run.py

closed 2019-05-29 by yguel

adam-urbanczyk, 2019-05-31:
Sorry to hear that it required so many workarounds. It is weird though that conda did not install all needed deps. Are you sure that you activated the conda environment before calling `python run.py`?

yguel, 2019-05-31:
Oh yes, sure, I activated the conda environment. But it works very nicely now, cq-editor too. And I am so happy to be able to use cadquery appart from freecad that it was worth the effort.
Thanks a lot for your answers, they helped a lot.