CadQuery/cadquery #133: Is the mode "combine=True" of function union correctly implemented?

bernhard-42, 2019-05-26:
I looked at `union` and the way `combine=True` works:
```
def union(...)
...
        if combine and solidRef is not None:
            r = solidRef.fuse(newS)
            solidRef.wrapped = newS.wrapped
...
```

Maybe I don't understand the mechanism, but shouldn't it be

```
        if combine and solidRef is not None:
            r = solidRef.fuse(newS)
            solidRef.wrapped = r.wrapped
```

so that `r` (the result) and `solidRef` (the original) point to the new object?

jmwright, 2019-05-29:
@bernhard-42 Well, intuitively what you're saying makes sense. @dcowden do you remember any particular reason it was done this way?

dcowden, 2019-05-30:
Actually I think I do.

In the original cq design and code, one goal was to ensure that operations that made modifications to a solid updated the original reference. The idea was that  references to a solid would be automatically updated to reflect the latest operations, without the need to continually re-assert the values.

In retrospect this decision was not wise. It produced a lot of confusion to save very little in the fluent api.

bernhard-42, 2019-05-30:
@dcowden That is in line what I also thought. However, then in case of union, solidRef.wrapped should definitely be r.wrapped. Currently the original points to the solid that is the union of the parameters of the union call and not to the end result.

@adam-urbanczyk With 2.0 there is a chance to make breaking API changes. How about changing all default parameters combine=True to False?
Or even better, get rid of the combine and clean up the API? (I am currently building a replay function for the fluent API and the combine feature is a major hasste.)

dcowden, 2019-05-30:
I would definitely support getting rid of combine in 2.0.  I think the simplification of the code would be quite significant, and we would give up very little.

There have been tons of questions over the years around it.

adam-urbanczyk, 2019-05-30:
@bernhard-42 agreed about always keeping the history of all operations. I need it in the CQ stack inspector in the CQ-editor. I think some of them were already modified to avoid in-place modifications of the wrapped object. 

Regarding combine it does make sense for some operations (e.g extrude) and I would like to keep it for such cases. For Boolean ops it can be indeed removed.

bernhard-42, 2019-05-31:
Need to better understand the advantages of combine for extrude et al. Will be back at my computer on Sunday to do some tests with extrude and understand how this combine is different from the combine of intersect and union.

adam-urbanczyk, 2019-05-31:
Here is an example:
```python
res1 = cq.Workplane('XY').box(3,3,1).faces('>Z').circle(1).extrude(1,combine=False)
res2 = cq.Workplane('XY').moveTo(0,4).box(3,3,1).faces('>Z').circle(1).extrude(1,combine=True)
```
[image]

bernhard-42, 2019-06-08:
@adam-urbanczyk 
I agree with your comment about combine for extrude. However, I would prefer to see the default for combine being `False`.

Let's look at the two examples:
```python
res = (cq.Workplane('XY')
         .circle(10).extrude(10))
```
and 
```python
res = (cq.Workplane('XY').box(30,30,10).faces('>Z')
         .circle(10).extrude(10))
```
While looking similar, the second line returns different solids for the two examples. So you need to have the implicit knowledge that extrude involves 1) the wire to extrude and 2) the underlying solid (just that in the first case the underlying solid is empty - with "empty solid" being hard to grasp). 

I would prefer to make it explicit, i.e. the default for `combine` is `False`

If I now read a line like
```python
res = (cq.Workplane('XY').box(30,30,10).faces('>Z')
         .circle(10).extrude(10, combine=True))
```
it is immediately clear that I have to extend the context of extrude from the wire to the box.

I am sure that there are good arguments for the current behaviour and maybe it is more of a philosophical question, but I thought to bring it up ;-)

adam-urbanczyk, 2019-06-09:
I'm afraid I don't really agree with your proposal @bernhard-42 . The typical use case is that you do want to combine extruded solids with the base solid. What is more, current default behavior follows the behavior of CAD programs.

bernhard-42, 2019-06-09:
no worries, it is not a real issue for me. I just wanted to at least discuss when you are currently cleaning up the combine flags.

bernhard-42, 2019-06-09:
And btw. PR #143 solved many of my current problems with my replay feature. So thanks a lot for implementing it!

PR #143 (2019-06-10): Remove the combine arg for bool ops and get rid of side-effects
This PR will resolve #133 

I removed the combine keyword argument from union, intersect and cut. This means that those functions never have side-effects, new cq stack object is always created.
closed 2019-06-10 by adam-urbanczyk