CadQuery/cadquery #134: cutThruAll produces incorrect results for offset workplane

mgreminger, 2019-05-30:
cutThruAll produces incorrect results for an offset workplane and the positive parameter has no effect. cutBlind produces the correct results for an offset workplane. See the examples below to reproduce the issue.

cutBlind produces the correct results for an offset workplane:

**Positive blind cut:** 
```
import cadquery as cq
test = cq.Workplane('XY').rect(100,100).extrude(100)
test = test.faces("<Z").workplane(offset=50,invert=True).rect(10,10).cutBlind(50)
```
[image]

**Negative blind cut:**
```
import cadquery as cq
test = cq.Workplane('XY').rect(100,100).extrude(100)
test = test.faces("<Z").workplane(offset=50,invert=True).rect(10,10).cutBlind(-50)
```
[image]

cutThruAll produces incorrect results for the same workplane:

**Positive thru-all cut:**
```
import cadquery as cq
test = cq.Workplane('XY').rect(100,100).extrude(100)
test = test.faces("<Z").workplane(offset=50,invert=True).rect(10,10).cutThruAll(positive=True)
```
[image]

**Negative thru-all cut:**
```
import cadquery as cq
test = cq.Workplane('XY').rect(100,100).extrude(100)
test = test.faces("<Z").workplane(offset=50,invert=True).rect(10,10).cutThruAll(positive=False)
```
[image]

This issue is most likely related to the face finding algorithm referenced here: https://github.com/CadQuery/cadquery/pull/113#issuecomment-469850786 . There are likely issues with this case since the model doesn't have a face adjacent to the offset workplane and ends up using the top or bottom face instead. The positive input argument is not referenced in the implementation of the method which explains why it has no effect.

adam-urbanczyk, 2019-05-30:
I am not sure that I understand the issue. cutThruAll cuts through everything and that is what you get.

 The positive keyword arg should is probably related to previous implementation and should be removed.

mgreminger, 2019-05-30:
Thanks, got it. Doc string implied a one direction cut, I've submitted a pull request to fix this (#135).

adam-urbanczyk, 2019-05-31:
@mgreminger  thanks for the PR! Regarding this issue, are you limited by the currently available functionality? If so, we could expose more operations implemented in the OCC kernel:

Method | Description
-- | --
Perform(Height) | The resulting prism is of the given length.
Perform(Until) | The prism is defined between the position of the base and the given face.
Perform(From, Until) | The prism is defined between the two faces From and Until.
PerformUntilEnd() | The prism is semi-infinite, limited by the actual position of the base.
PerformFromEnd(Until) | The prism is semi-infinite, limited by the face Until.
PerformThruAll() | The  prism is infinite. In the case of a depression, the result is similar to  a cut with an infinite prism. In the case of a protrusion, infinite  parts are not kept in the result.
(from: https://www.opencascade.com/doc/occt-6.9.1/overview/html/occt_user_guides__modeling_algos.html#occt_modalg_9)

currenty `PerformThruAll()` is used and you were probably after `PerformUntiEnd()`.

mgreminger, 2019-05-31:
Thanks, @adam-urbanczyk. Yes, PerformUtilEnd() is what I was expecting. The current functionality doesn't limit what I am trying to do, cutBlind() works for my needs. I just got confused about what was happening when I was getting a two directional cut. The doc string update and removing the positive argument will make the intent of cutThruAll() clear.  However, exposing PerformUntilEnd() in some way would probably be useful in the long run for cases like mine where the workplane is offset from a face since I'll have to use a large cut distance in cutBlind() to make sure the single direction cut goes through all of the material.

commit 86f688a6 (2019-06-07): Merge pull request #135 from mgreminger/issue_134

Update cutThruAll to remove unused positive argument (fixes #134)
closed 2019-06-07 by adam-urbanczyk