CadQuery/cadquery #136: Will there be pip support?

grbd, 2019-06-01:
I'm not sure if this is the best place to ask this question
but I've noticed with pythonOCC they only seem to support Conda not pip
pip support seams to have been removed a while back

Since cadquery is dependent on pythonOCC this means pip support might not be possible in the future?

jmwright, 2019-06-01:
Welcome @grbd

@adam_urbanczyk will know better than me, but I think it would be a significant undertaking to support installation via pip. We'd certainly take contributions in this regard, but I'm not sure when/if it will get done with our current resources.

adam-urbanczyk, 2019-06-01:
@grbd indeed no plans to support pip extensively. You can install cadquery with pip but it will not take care of the dependencies like pythonocc or oce. I don't think it would be practical/possible to make a oce pip package.

dcowden, 2019-06-01:
I think we should support a standard docker container. That makes a simple
to run, one step install that will always work.

The container should allow exectuting a script at the command line, or
running in daemon mode existing an http endpoint that can serve up a
gui(electron).

We need two containers. One should be a full conda that will be rather
large. The other should be a miniconda with just the minimal stuff
installed.

A kubernetes manifest should be provided also (Mac now comes with a great
minikube install that would make it super easy to run as a k8s manifest)

grbd, 2019-06-01:
I've been warming up to the idea of conda after reading
https://jakevdp.github.io/blog/2016/08/25/conda-myths-and-misconceptions/

adam-urbanczyk, 2019-06-05:
@grbd did conda work for you in the end?

grbd, 2019-06-05:
Yep I managed to get it to work
I'll close this down since it's really depending on pythonOCC as far as pip is concerned

closed 2019-06-05 by grbd