CadQuery/cadquery #137: Mesh / polyhedron support?

grbd, 2019-06-02:
Hi,
one of the things I'm interested in is the generation of gears / racks etc
I figured I'd have a go at trying to convert some of the openscad scripts into python using cadquery. (so far I've got the latest 2.0 running with CQ-editor)

https://github.com/chrisspen/gears/blob/master/gears.scad

However it looks like these use something called a polyhedron which is basically a list of points
then a list of which points make a face to create the solid
Stricly speaking this isn't parametric cad, it's more of a mesh so may fall outside the scope

Is there a way to do this currently using cadquery?
I've been digging through the docs but couldn't find an example

I suppose there are better ways to do it from a parametric point of view, but it would involve working out how all the math works.

dcowden, 2019-06-02:
hi @grbd!!

The approach in CQ would be to create the gear geometry using lines and arcs-- and then (probably) extrude into a gear

You might find this example useful:

https://cadquery.readthedocs.io/en/latest/examples.html#cycloidal-gear

You definitely would model a gear in python using the correct curves-- not just an array of points.

For a good gear library, you probably can't escape doing the actual math for the involute profiles-- that's the math that's there already in gears.scad.

the real power of CQ is that it is based on python.  I would hope someone has already written python versions of spur gear code ( https://github.com/heartworm/py_gear_gen  for example )

grbd, 2019-06-02:
thanks for the info

dcowden, 2019-06-02:
You are very welcome! Thanks for trying our cadquery! A spur gear module would be amazing!

closed 2019-06-02 by grbd