CadQuery/cadquery #139: Intermittent missing faces for face selectors

mgreminger, 2019-06-05:
**Description**
Sometimes a face or two is missing from a face selector call. It doesn't happen very often but it does lead to errors that seem to occur randomly. This sometimes leads to the test suite failing on code that will work the next time you run the test suite (see example [here](https://travis-ci.org/CadQuery/cadquery/jobs/536938386#L1384) for a test failure for code that passes most of the time). These errors occur on the tests that rely on returning the correct number of faces.

**How to Reproduce**
The code below reliably reproduces this issue by running one of the tests 1000 times. This code is based on the `testRectArray` test in `TestCadQuery.py`.

```python
import cadquery as cq

tally = {}
n = 1000

for i in range(n):
    NUMX = 3
    NUMY = 3
    s = cq.Workplane("XY").box(40, 40, 5, centered=(True, True, True)).faces(
           ">Z").workplane().rarray(8.0, 8.0, NUMX, NUMY, True).circle(2.0).extrude(2.0)

    count = tally.get(s.faces().size(),0) + 1
    tally[s.faces().size()] = count
    
for key, value in tally.items():
    print(f"{key} faces is returned in {value} out of {n} runs")
```
*The above produces the following output (the numbers will vary):*
```
24 faces is returned in 994 out of 1000 runs
23 faces is returned in 5 out of 1000 runs
22 faces is returned in 1 out of 1000 runs
```
**Likely Cause**
Occasionally the hash codes returned for the opencascade objects are repeating leading to some objects being left out when they are accumulated in the [_collectProperty method of CQ](https://github.com/CadQuery/cadquery/blob/9aaf110c59837e5b8915f6065950311c2a9ca478/cadquery/cq.py#L84-L110). This is occurring because the value for `HASH_CODE_MAX` is too small. It is defined here in shapes.py:
https://github.com/CadQuery/cadquery/blob/9aaf110c59837e5b8915f6065950311c2a9ca478/cadquery/occ_impl/shapes.py#L117
This occurs because of the way opencascade computes the hashes. The relevant opencascade code is [here](https://github.com/tpaviot/oce/blob/9a62aead6daa5a4adb3b34161c70810e4d641e2c/src/TopoDS/TopoDS_Shape.cxx#L24-L31). It's doing a bitwise XOR between a hash of the objects pointer and a hash of the object's location and then computing the modulus between that hash and the upper limit that we're setting using `HASH_CODE_MAX`. If `HASH_CODE_MAX` is too small, we'll get the occasional collisions that we're seeing in the test suite failures. 

**Possible Solution**
Increasing the size of `HASH_CODE_MAX` will decrease the probability of these collisions. I'm not sure if it's possible to completely eliminate the chance of hash collisions given how OpenCascade computes it's hashes, however, by increasing `HASH_CODE_MAX`, I've been able to eliminate the hash collisions for the above test case (even when increasing n to 100,000) so we'll see an immediate improvement and this should eliminate the random test suite failures that we see. Both pythonocc and freecad use a value of 2147483647 for `HASH_CODE_MAX` (the relevant pythonocc code is [here ](https://github.com/tpaviot/pythonocc-core/blob/2ccaefb7cc1e9b56093ec9f7a7fd7ae7c89d4f23/src/SWIG_files/wrapper/TopoDS.i#L875-L881) and the relevant freecad code is [here](https://github.com/FreeCAD/FreeCAD/blob/a56658489bc008a972aeac37cc8eeff192090ef4/src/Mod/Part/App/TopoShapePyImp.cpp#L1321-L1326) where INT_MAX is 2147483647 for 4 byte integers). This is the largest int allowed for 4 byte integers, if `HASH_CODE_MAX` is any larger, it crashes. We don't have to worry about less that 4 byte integers since pythonocc hard codes this number so pythonocc will only compile with 4 byte or larger integers and `HASH_CODE_MAX` is specific to the opencascade implementation of `Shapes`. Also, since pythonocc uses the same number, the output of `HashCode()` will now match `__hash__()` provided by the pythonocc bindings.

This does require us to have a hard coded a magic number since Python 3 doesn't provide sys.maxint anymore due to python 3 integers being unbounded. We could use the following code to compute MAX_INT, but I think this is worse that just using the number:
```python
import struct
platform_c_maxint = 2 ** (struct.Struct('i').size * 8 - 1) - 1
```

dcowden, 2019-06-05:
Wow this is amazing sleuthing! Thanks for running this down! I agree with your recommendation

adam-urbanczyk, 2019-06-05:
@mgreminger  this code smell is on me. Thanks for tracking this down!

mgreminger, 2019-06-05:
No problem, glad we could get to the bottom of it. The real issue is the way that OpenCascade assigns hash codes. It seems that it relies on chance to avoid collisions. 

Thanks a lot to all of the team for the cadquery project. You all have made OpenCascade accessible in a way that's a joy to use, which is quite remarkable feat!

closed 2019-06-05 by mgreminger