CadQuery/cadquery #147: polyline is not correct, wrong position of first point

yguel, 2019-06-11:
Look at the code below:

<code>
import cadquery as cq
d0 = 17.25
h1 = 25

tri_z_min = 0
tri_O = (4,tri_z_min)
tri_A = (-2*d0,tri_z_min)
tri_B = (tri_A[0],2.25*h1)

tri_plane = cq.Plane( cq.Vector(0,d0,0),cq.Vector(1,0,0),cq.Vector(0,-1,0) )

pts = [ tri_O, tri_A, tri_B ]
cut_block = cq.Workplane(tri_plane).polyline(pts).close().extrude(2*d0)

show_object(cut_block)
</code>

The prism rightmost edge should have x coordinates equal to 4, but it is 0. It looks like the Workplane coordinate system and the polyline definition have been messed-up.

adam-urbanczyk, 2019-06-16:
It is an interesting issue. If you inspect the stack in CQ-editor it seems that the `close()` command does not work properly:
[image]

adam-urbanczyk, 2019-06-20:
This is the root cause:
https://github.com/CadQuery/cadquery/blob/ef0808f4104e7c82da5df0e66355da0f3bbd24af/cadquery/cq.py#L1915

I am not sure if this is a bug or a feature, any thoughts @dcowden @jmwright ?

@yguel  workaround for now: move the center to the first point and then us `polyline`.

jmwright, 2019-06-21:
I think that this has caused confusion before, so I'm ok changing it as long as all the tests still pass. Hopefully there's not some edge case that this will break that are not covered in the tests.

PR #156 (2019-06-25): Change default polyline and spline behavior
This will resolve #147
closed 2019-06-25 by adam-urbanczyk