CadQuery/cadquery #148: Fluent API does not allow to easily extrude an existing face

adam-urbanczyk, 2019-06-12:
It is possible with the following (not so fluent) workaround:

```python
cap_m = cap.cut(cut_block,combine=False).faces('<Z').wires()
cap_m.ctx.pendingWires = cap_m.vals()
cap_m = cap_m.extrude(-h3,combine=True)
```

This could be solved by e.g. adding an additional function like `toPendingWires()` which would append current `vals()` to the `pendingWires` field.

adam-urbanczyk, 2019-06-13:
I think you are interested in this issue @yguel

bernhard-42, 2019-07-07:
@adam-urbanczyk  For me `ctx.pendingWires` looks like an internal implementation object (it does e.g. not appear in the docs), so not sure naming something `toPendingWires` is very intuitively.

I would see two other options here:
1) Extend `wires` to `wires(self, selector=None, forExtrusion=False)`
If `forExtrusion==True`, the wire would be added to the pending wires context else not.
2) Allow directly `anyface.extrude` and if self is one or more faces automatically do the pending wires trick

Not sure what my favourite is, 1) is consistent with the fact that extrude works on wires and makes this explicit, however makes `wires` slightly different to the behaviour of `edges` and `faces`. 2) on the other side is simply shorter which is also nice for a commonly used pattern.

Hatatister, 2019-08-08:
Hello,
I have started using cadquery recently. First I have to say that it is a great project, which is so much more useful than the lowlevel pythonocc api.

Now back to the topic:
@adam-urbanczyk 
I had the same issue trying to extrude a face directly. I solved it by adding a function.
I think `toPendingWires` is not obvious for someone who does not know the source code. I called it `faceBoundary`

To me bernhard-42's 2. suggestion makes sense. A user would expect that an extrusion of a face would result in a solid

r4dr3fr4d, 2020-02-17:
@bernhard-42 the only issue I see with those instead of toPendingWires() is that we'd have to create such methods for every use of pendingWires besides extrude - such as loft.    toPendingWires() might be cleaner

lalebarde, 2020-07-05:
Hi all, I fully agree with bernhard-42's 2. This would be very useful to have.

adam-urbanczyk, 2020-07-05:
This is solved by #351. The relevant method is `toPending`

closed 2020-07-05 by adam-urbanczyk

lalebarde, 2020-07-05:
Hi Adam, I am trying to reuse your OP example, but my context starts differently and I have an error:

    decoupe = cq.Workplane("XY").rect(10, 20, centered=True).extrude(100)
    decoupe = decoupe('<Y').wires()
    decoupe.ctx.pendingWires = decoupe.vals()
    decoupe = decoupe.extrude(-100,combine=True)

The error comes at line 2:

    TypeError: 'Worplane' object is not callable

ulikoehler, 2024-05-18:
For everyone finding this ticket and trying to extrude a face, here's how you can do it:
```
result = face_to_extrude.extrude(2, taper=45)
```

Full example:
```python
import cadquery as cq

box = cq.Workplane("XY").box(1, 1, 1)

face_to_extrude = box.faces(">Z")

result = face_to_extrude.extrude(2, taper=45)
```
[image]

From my blog: https://techoverflow.net/2024/05/18/how-to-extrude-a-face-in-cadquery/

PR #351 (2020-06-02): Add DXF import to the OCP branch
Preliminary support for DXF importing. Entities are converted to faces. So far only lines, polylines, arcs and circles are supported.

NB: to enable further processing `toPending` is also implemented in this PR, so it will resolve #148 

Example usage:
```python
res = importers.importDXF('gear.dxf',1e-3).wires().toPending().extrude(5)
```

[image]