CadQuery/cadquery #149: Workplane() resets currently selected objects

adam-urbanczyk, 2019-06-13:
The following code does not work
```python
import cadquery as cq
result = cq.Workplane("XY").rect(10.0,10.0).extrude(10).faces("<Y")\
    .workplane().vertices('<XZ').circle(1).extrude(10)
```
I think it should -- it is quite clear what the intent if the user is here. This is also related to #24

Jojain, 2022-01-25:
We agree that the expected behaviour would be equal to this : 
```python
import cadquery as cq
result = cq.Workplane("XY").rect(10.0,10.0).extrude(10).faces("<Y")\
    .workplane().end(1).vertices('<XZ').circle(1).extrude(10)```
I have been bothered several time by this and while the `end` method solves the problem, it's super unnatural and definitely not novice friendly.

If this is still expected to be solved I will try to bring a PR

adam-urbanczyk, 2022-01-26:
I think, that's it indeed.

adam-urbanczyk, 2022-01-26:
Or maybe:
```python
s.objects = self.objects[:]
```

Jojain, 2022-01-26:
I have tried my solution, it breaks some tests, I have quickly looked, some tests could be rewritten, some need a bit more investigation why they break.

adam-urbanczyk, 2022-01-26:
Feel free to open a WIP PR and we can take a look together.